cyberbotics/webots #2848: RoboCup Stadium is aligned "wrongly"

jgueldenstein, 2021-03-11:
**Describe the Bug**
The RoboCup stadium is aligned with the y-axis pointing towards the goal. In the team communication protocol, it is defined that the x-axis points towards the goal (https://github.com/RoboCup-Humanoid-TC/RobocupProtocol/blob/1ab5585c33a83c2df9e2513995c60bc8860e67f5/robocup.proto#L39). That is why we would like to rotate it

The rotation of the field can not be changed for some reason, probably because it is autogenerated.

**Steps to Reproduce**
1. Open the RobocupSoccerField.proto field.
2. Enable "show coordinate system" (CTRL+F1)
3. Change the rotation field of the SoccerField node in the scene tree.

**Expected behavior**
x-axis points towards one of the goals or the world can be rotated.

**Screenshots**
[image]
[image]

omichel, 2021-03-12:
Thank you for the feedback. We will fix this as soon as possible.

omichel, 2021-03-16:
Fixed in #2851.

closed 2021-03-16 by omichel

jgueldenstein, 2021-03-16:
@omichel thank you for taking care of this so quickly

PR #2851 (2021-03-16): Fixed translation/rotation field of the RobocupSoccerField proto
Fixes #2848:
- the translation and rotation parameters of the proto were not linked to the proto definition.
- the X direction was not aligned with the goals by default as it should be.
- the ads on the stadium background were mirrored.