cyberbotics/webots #2855: Pedestrian crossing does not appear above a straight road

nelsondmmg, 2021-03-15:
**Describe the Bug**
Pedestrian crossing does not appear above a straight road

**Steps to Reproduce**
Adding the crossing by the add an object

**Expected behavior**
The crossing should be reproduced above the road

**Screenshots**
[image]

**System**
 - Webots R2021a revision 1
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce RTX 2060 mobile

stefaniapedrazzi, 2021-03-15:
Note that this seems an issue due to the semi-transparent PNG image.
If the texture URL is changed to point to a JPG texture, then it works correctly and the texture is displayed in the scene.

nelsondmmg, 2021-03-15:
The crossing was displayed correctly? Because I converted the crossing.png into a jpg and then changed the PROTO to point to the jpg version and this is the result:
[image]

stefaniapedrazzi, 2021-03-15:
I simply tested changing the texture and with JPG something was displayed but honestly didn't check if the JPG was displayed correctly.
So it could simply be that the texture mapping is wrong.

stefaniapedrazzi, 2021-03-15:
The `PedestrianCrossing` PROTO has been broken by #2364.

stefaniapedrazzi, 2021-03-15:
@nelsondmmg Thank you for reporting this issue.
The bug has been fixed in  #2857.
You can apply the changes to your local file or wait for the next nightly build of R2021a-rev1.

PR #2857 (2021-03-15): Fix PedestrianCrossing PROTO
Fix #2855: update the `PedestrianCrossing` PROTO following #2364.
closed 2021-03-15 by stefaniapedrazzi