cyberbotics/webots #2858: Clean exit of the Webots process when the (a specific) Supervisor's controller terminates

mrshu, 2021-03-15:
**Is your feature request related to a problem? Please describe.**

While playing with the RoboCupJunior Soccer Sim (essentially a few controllers + Webots; https://github.com/RoboCupJuniorTC/rcj-soccer-sim/), we ran more than a few times into a situation when the Supervisor ran into an exception and the whole simulation still continued, despite being in a state which was not really helpful. This was especially the case when trying to generate a video of the simulation -- we've lost quite some time that way.

Hence the request: being able to cleanly exit the Webots process when the Supervisr's controller terminates.

**Describe the solution you'd like**
A command-line flag which would make the Webots binary terminate as soon as the (a specific/any) Supervisor's controller terminates.

Alternatively, for the specific use case I have in mind it would be enough for the Webots binary to terminate whenever any controller terminates.

**Describe alternatives you've considered**
Since the code we're running in our particular use case is all in Python, we could potentially wrap everything in a large try/except block and then cleanly exit the Webots environment when we receive some exception. That opens a different can of worms though.

----------------

I am not sure how to go about implementing something like this but if you could give me some pointers, I'd be happy to send a PR.

Thanks!

omichel, 2021-03-16:
Why don't you use [wb_supervisor_simulation_quit](https://www.cyberbotics.com/doc/reference/supervisor#wb_supervisor_simulation_quit) in your supervisor? That would fix the problem.

closed 2021-03-16 by omichel

mrshu, 2021-03-16:
Thanks for your suggestion @omichel!

Just to check that I understand it completely, since the controller is written in Python, you'd suggest we use something like [`atexit`](https://docs.python.org/3/library/atexit.html) and call `wb_supervisor_simulation_quit` in its handler? I guess that would probably work and would not require any further updates to Webots.

Thanks again for all your help!

omichel, 2021-03-16:
Yes, that's one possibility. Another one would be to call this function at the end of your Python program before quitting. If your program may fail for some reason, you should probably use try/catch constructs to call the `Supervisor.quit()` method in the catch.