cyberbotics/webots #2859: Handling infinite loops in controllers

mrshu, 2021-03-15:
**Is your feature request related to a problem? Please describe.**

This is one of the few things we've ran into while playing with the RoboCupJunior Soccer Sim (essentially a few controllers + Webots; https://github.com/RoboCupJuniorTC/rcj-soccer-sim/). As it was kids from all over the world writing code and submitting it for our "competition", there was a chance at least some of the contributed controllers would end up in infinite loops. This is undesirable, as an infinite loop of one of the controllers freezes the whole Webots process.

We were hence wondering where there was a simple way of perhaps detecting that this is happening and then stopping just the process that belongs to the controller in the infinite loop.

**Describe the solution you'd like**
Terminate the process which is in the infinite loop.

**Describe alternatives you've considered**
We also considered doing sort of a "heartbeat", in which each controller "touches" a specific file and an outside process detects when this does not happen for a longer periods of time (say a couple of seconds). But it would be great if this was supported by Webots itself.

Furthermore, this would obviously be a non-issue with the ROS integration which pushed the problem to a completely different level of abstraction but then again, the ROS integration brings a totally different set of issues as well.

-------

This is something I am not sure is even possible but I still wanted to ask what your thoughts on something like this would be.

Thanks a ton!

omichel, 2021-03-16:
There is a simple solution to this problem: [`Robot { synchronization FALSE }`](https://cyberbotics.com/doc/reference/robot).

closed 2021-03-16 by omichel

mrshu, 2021-03-16:
Thank you very much for your fast reply @omichel -- that's exactly what we were looking for!