cyberbotics/webots #2860: Configurable `PYTHONPATH` to allow for code sharing between controllers

mrshu, 2021-03-16:
**Is your feature request related to a problem? Please describe.**

This is another (hopefully last) issue we've ran into when using Webots as the platform for RoboCupJunior Soccer Sim (https://github.com/RoboCupJuniorTC/rcj-soccer-sim/).

In it, two teams (of kids) submit code for three robots each and play a game of soccer in the environment which is basically a verbatim copy of the `soccer` demo distributed with Webots.

Since they submit code for three robots, much of it is shared. This sadly doesn't play very nicely with the concept of controllers as Webots sees them, as it expects each controller (robot) to be self-contained in a separate directory. We've tried to work around it by [suggesting they update the `PYTHONPATH`  from their code](https://robocupjuniortc.github.io/rcj-soccer-sim/how_to/#importing-shared-code), something like the following:

```python
import sys
from pathlib import Path
sys.path.append(str(Path('.').absolute().parent))
from robot1 import utils
```

but this brings a ton of other problems, like the requirement for unique controller names.

We were therefore wondering whether you'd be open to making the `PYTHONPATH` with which the controller processes are executed configurable, say from within the World file. The implementation ought not to be very difficult either, as something similar is already happening with `addToPathEnvironmentVariable` 

https://github.com/cyberbotics/webots/blob/785ece7cef9d322af09cca786bb66f4144595e56/src/webots/control/WbController.cpp#L514-L528

**Describe the solution you'd like**
A way of specifying custom paths to be added to `PYTHONPATH` which is specified for the Python Controller processes.

**Describe alternatives you've considered**
- _symlinks_: these would probably be the perfect solution but it is sadly not really platform-independent
- _updating `PYTHONPATH` from within the controller code_: this is our current solution but it is quite clunky and not really "Pythonic"

-------------------------------

As I mentioned before, I'd be happy to try to put something like this together if you believe such addition would make sense in Webots, I would be happy to try to put it together.

Thanks!

cc @Adman

omichel, 2021-03-16:
Did you try to set the `PYTHONPATH` in the [runtime.ini](https://cyberbotics.com/doc/guide/controller-programming#environment-variables) file of the controller folder?

In robot soccer competition, it can also be simpler to request that the competitors provide a single controller that will be used for all robots (each robot of the same team will run their own instance of the same controller). From the controller, it is possible to [retrieve the robot name](https://cyberbotics.com/doc/reference/robot#wb_robot_get_name) and behave differently if the robot name is "red goal keeper" or "blue attacker 2", etc. This can make things way simpler to handle redundancy of code in controllers for different players.

closed 2021-03-16 by omichel

mrshu, 2021-03-20:
Thanks @omichel, in the end we've opted for using the single controller approach, which, as you mention, is both much simpler and cleaner, especially since the robot name has already been used  and well documented in our materials.

Thanks again for all your help!