cyberbotics/webots #2869: Webots Documentation broken

aschokking, 2021-03-17:
**Documentation page(s) concerned**
Every page seems to have an error rendering right now in latest chrome, eg: https://cyberbotics.com/doc/reference/hingejoint

**What is the problem with the current documentation?**
Error rendering
[image]

Tried in incognito as well so it's not a local caching issue.

**Describe how you would fix it**
Something fundamental must have gone wrong.

omichel, 2021-03-17:
Well, I just tried and it worked for me...
May I ask you to try the following URLs and report and success or failure on them?
- https://cyberbotics.com/doc/reference/hingejoint
- https://cyberbotics.com/doc/reference/hingejoint?version=76f58481c802da8919b49e2547e5e133f8b345e4
- https://cyberbotics.com/doc/reference/hingejoint?version=R2021a
- https://cyberbotics.com/doc/reference/hingejoint?version=released
- https://cyberbotics.com/doc/reference/hingejoint?version=master
- https://cyberbotics.com/doc/reference/hingejoint?version=develop

omichel, 2021-03-18:
I actually applied a quick and dirty patch to fix this temporarily.
We need a better solution to this, this is why I am reopening the issue.

rokn, 2021-03-20:
I think I'm still hitting a problem. It's on every page of the documentation for me and it looks like this:
[image]
The whole text of every page is pushed to the left like this. It's like this on links like this:
https://cyberbotics.com/doc/reference/hingejoint
Or this:
https://cyberbotics.com/doc/reference/hingejoint?version=released
The other ones you mentioned work correctly but on this release versions I'm getting the same problem across every docs page.

aschokking, 2021-03-20:
It looks like a css loading issue, the pages you linked to load fine for me
at least right now.

rokn, 2021-03-20:
Yup it tries to load webots-doc.css two times but both are failing with 403.
Here is the response:
[image]
And here is the request:
[image]

jasonswearingen, 2021-03-20:
site is broken because ```cdn.jsdeliver.net``` is blocking the content.   Offending resource is [https://cdn.jsdelivr.net/gh/cyberbotics/webots@released/docs/css/webots-doc.css](https://cdn.jsdelivr.net/gh/cyberbotics/webots@released/docs/css/webots-doc.css)

click through to see the error text that is returned instead of the css:  ```Package size exceeded the configured limit of 50 MB. Try https://github.com/cyberbotics/webots/tree/released/docs/css/webots-doc.css instead.```

rokn, 2021-03-21:
It works for me now.