cyberbotics/webots #2884: Integrate `QCommandLineParser`

lukicdarkoo, 2021-03-18:
Since Qt 5.2 a `QCommandLineParser` class is available:
https://doc.qt.io/qt-5/qcommandlineparser.html

We should integrate it to make the code cleaner and to expand the CLI capabilities.

The new parser should be added here:
https://github.com/cyberbotics/webots/blob/55c40cdb536322ad56a51414ce3fb653df31d759/src/webots/gui/WbGuiApplication.cpp#L192

and it should look like something like this:
```c++
  QCommandLineParser parser;
  parser.setOptionsAfterPositionalArgumentsMode(QCommandLineParser::ParseAsOptions);
  parser.addPositionalArgument("subcommand",
    "subcommand1\n"
    "subcommand2\n" // ....
  );
  parser.parse(arguments());
  QStringList parserArgs = parser.positionalArguments();

  if (!parserArgs.isEmpty()) {
    // Parse subcommands
    const QString subCommand = parserArgs.first();
    if (subCommand == "subcommand1") {
      parserArgs.pop_front();
      subProgram(parserArgs);
    }
  }

  void subProgram(QStringList parserArgs) {
    QCommandLineParser parser;
    parser.setOptionsAfterPositionalArgumentsMode(QCommandLineParser::ParseAsOptions);
    parser.addPositionalArgument("myparam", "Some description.");
  }
```

bkpcoding, 2021-03-20:
Working on it will be able to send PR by today

bkpcoding, 2021-03-20:
So, I am a little confused here, can someone help, as far as I understood, 
-  parser.positionalArguments() returns a list of positional arguments (these are the arguments that do not start with "--" or "-"
- But the function: https://github.com/cyberbotics/webots/blob/55c40cdb536322ad56a51414ce3fb653df31d759/src/webots/gui/WbGuiApplication.cpp#L192  has only options, and I don't see any subcommands, can you help me understand what do the parserArgs() list contains.

bkpcoding, 2021-03-20:
Or am I going in the right direction?
```
include<QCommandLineParser>
void WbGuiApplication::parseArguments() {
  // faster when copied according to Qt's doc
  QCommandLineParser parser;
  parser.setOptionsAfterPositionalArgumentsMode(QCommandLineParser::ParseAsOptions);
  parser.addPositionalArguments("subcommand"
                                "minimize\n"
                                "fullscreen\n"
                                "mode\n"
                                "no-rendering\n"
                                "help\n"
                                "sysinfo\n"
                                "version\n"
                                "batch\n"
                                "update-proton-cache\n"
                                "update-world\n"
                                "stream\n"
                                "stdout\n"
                                "stderr\n" //..
                                );
parser.parse(arguments());
QStringList parseArgs = parser.positionalArguments();

while(!parserArgs.isEmpty()){
    const Qstring subCommand = parserArgs.first();
    if(subCommand == "minimize"){
        parserArgs.pop_front();
        mshouldMinimize = true;
    }
    else if(subCommand == "fullscreen"){
        parserArgs.pop_front();
        mShouldStartFullscreen = true;
    }
    //.....
}
    parser.clearPositionalArguments();
```

lukicdarkoo, 2021-03-20:
> has only options, and I don't see any subcommands, can you help me understand what do the parserArgs() list contains.

The subcommands will be added later. The first step is to integrate the `QCommandLineParser` while keeping the same API. From the user's perspective, everything should be the same. Adding subcommands like `stdout` doesn't make any sense, it should be flag as it is now.

Don't blindly follow the example given in the issue. Remember, the objective is to integrate the `QCommandLineParser` while keeping the same API.

omichel, 2022-05-02:
After a draft implementation in #4427, it turns out that QCommandLineParser doesn't provide any added value over the previous code, because we had to override most of its useful features to keep the previous behavior and functionality. So I am closing this issue for now.

closed 2022-05-02 by omichel