cyberbotics/webots #2889: Suction cup simulation

carlosprados11, 2021-03-19:
**Describe the Bug**
I have successfully simulated a suction cup in Webots via Connectors (active the connector located in the suction cup, and passive the connectors of the environment that I have spread along the objects). However, for huge environments, the need to create a uniform distribution of Connectors causes the world's load to fail.

Is there another option to simulate a suction cup? Do you find a way to solve the problem in huge environments?

Actually, it would be similar to a magnet that stick to the iron plate.

**Expected behavior**
The suction cup to be attached to any surface.

**System**
 - Operating System: Ubuntu 20.04

lukicdarkoo, 2021-03-22:
Can you attach a world that throws the error?

omichel, 2021-03-22:
I believe the suction cup actuator should be added as a new basic node in Webots, instead of using connector nodes.
Basically, `SuctionCup` should be inspired from `Connector` except that it could connect (e.g., created a fixed joint) with any `Solid` node. @carlosprados11, if you feel like you would be able to implement it, we would be happy to review your PR about it. Please be sure to follow our [guidelines for adding a new device](https://github.com/cyberbotics/webots/wiki/Adding-New-Node-and-API-Function) in Webots.

lukicdarkoo, 2021-03-22:
> `SuctionCup` should be inspired from `Connector` except that it could connect (e.g., create a fixed joint) with any `Solid`

I would rather consider extending the `Connector` node to connect to any `Solid`, like adding an extra flag?

carlosprados11, 2021-03-22:
> > `SuctionCup` should be inspired from `Connector` except that it could connect (e.g., create a fixed joint) with any `Solid`
> 
> I would rather consider extending the `Connector` node to connect to any `Solid`, like adding an extra flag?

`Connector` right now needs two nodes to link. Maybe is interesting to create a new basic node that takes the `boundingObject` from other nodes to attach them according to the material properties of the surface and to the suction cup features.

omichel, 2021-03-22:
> I would rather consider extending the `Connector` node to connect to any `Solid`, like adding an extra flag?

Yes, that seems to be a good idea. However, I wouldn't add an extra flag, but rather add an extra type. In addition to "symmetric", "active" and "passive", I would add "suction cup".

> `Connector` right now needs two nodes to link. Maybe is interesting to create a new basic node that takes the `boundingObject` from other nodes to attach them according to the material properties of the surface and to the suction cup features.

I am not sure to understand what kind of new basic node you want to introduce here, but I believe it's not needed.

carlosprados11, 2021-04-08:
Do you know if it is possible to dynamically generate a Connector in a Solid? Perhaps it is a solution to the problem, generating the Connector at the point where the suction cup will stick.

omichel, 2021-04-13:
> Do you know if it is possible to dynamically generate a Connector in a Solid? Perhaps it is a solution to the problem, generating the Connector at the point where the suction cup will stick.

This doesn't seems to be an efficient way of implementing a suction cup. I would rather modify the `Connector` node as suggested earlier. To implement this, you will have to create an ODE fixed joint between the `Connector` and the `Solid` node that is being sucked, at the point of contact between them.

EnriqueRebollo, 2021-09-17:
Hello, I'm working with @carlosprados11 and taking over the development of the new component _SuctionCup_ for Webots.
I've been several days getting familiar with Webots and the different files related to the _Connector_ node, especially with the functions present in the file _WbConnector.cpp_. At this point, I'm stuck at the point of adapting the function _WbConnector *detectPresence()_ to get to detect the presence of a solid, as this one works by checking iteratively if the connectors present in the simulation (defined in the list _gConnectorList_) are close enough.
In the first place, I'm not sure if there's a list with all the solids present in the simulation. But if there is, I'm not sure this would work, because whereas the Connector has no significant volume and is more or less a unique point in the space, a solid has volume, and therefore we won't just need to know the distance from the _SuctionCup_ node to the _Solid_, but from the _SuctionCup_ to the closer part of the solid.
Therefore, how would you approach this? Do you think it is possible to keep a similar function structure or will it be necessary to think of a new approach?
Thanks in advance

omichel, 2021-09-22:
I would rather add an ODE sphere to the SuctionCup node and detect collision between this sphere and other solid objects in the world. If such a collision is detected, that would mean that the colliding solid is in vicinity of the suction cup and can be sucked, e.g., a fixed joint should be created between the colliding solid and the cup.

EnriqueRebollo, 2021-09-30:
Okay, so the approach for the connection would be:
Whenever a collision occurs between the Sphere and a Solid, and the SuctionCup is active, a joint will be dynamically created between the Sphere and the Solid, right?
In this case, is it possible in Webots to dynamically create joints during the simulation?

omichel, 2021-10-01:
> Okay, so the approach for the connection would be: Whenever a collision occurs between the Sphere and a Solid, and the SuctionCup is active, a joint will be dynamically created between the Sphere and the Solid, right?

Yes exactly.

> In this case, is it possible in Webots to dynamically create joints during the simulation?

Yes, a ODE fixed joint can be created dynamically, as it done currently in `WbConnector.cpp`.

closed 2021-11-15 by BenjaminDeleze