cyberbotics/webots #2891: Web simulation quick start instructions not working

stefaniapedrazzi, 2021-03-22:
As reported by a user the quick start instructions described [here](https://www.cyberbotics.com/doc/guide/web-simulation#quick-start) are not working: the streaming_viewer doesn't start the Webots instance.

Moreover there is an issues with the ports: the port number logged in the simulation and session `output.log` files doesn't match the ones printed in the console.
For example in the logs for default configuration I always get port (session: 1999, server:2000) but in the console the printed ports seems randomly chosen (for example 14603 and 14604 or  15175 and 1575).
But when restarting the server I get an error about addresses already used, so I would say that the wrong ones are the random values printed in the console.

anastasiapan, 2021-04-15:
Is this issue related only to the documentation? Because I would still like to setup my server to run a Webots instance and see it from my machine via browser streaming and I thought to use this guide. So if it is only in the documentation is it possible to point out to me the correct instructions?

stefaniapedrazzi, 2021-04-15:
It is just the approach described in the `Quick Start` section and using the `streaming_viewer` that doesn't work.

If you setup your website host instead of using the `streaming_viewer` code then it works correctly.
Here it is described hot to setup the host:
https://www.cyberbotics.com/doc/guide/web-simulation#website-host

stefaniapedrazzi, 2021-05-26:
Instructions should be checked again and fixed if needed.

BenjaminDeleze, 2021-07-13:
I followed the quick start guide and everything worked according to the documentation. I could start webots from the streaming_viewer/index.html

Concerning the ports: there are always set to 1999 and 2000 by default.
What is printed in the console, for example: `Started default session server (214865)` is not the port. It is the pid.

I am closing it for now, don't hesitate to reopen it if I missed something.

closed 2021-07-13 by BenjaminDeleze