cyberbotics/webots #2901: Webots crashes within 3 seconds of opening.

TheCottonMan, 2021-03-24:
Hi

I have just recently installed Webots (very excited) and the whole program crashes/closes within 3 seconds of it launching. I am unable to provide any further details as it is not open long enough.

I have uninstalled it many times and re-downloaded it, all to no avail.

I am running Windows 10 Pro. All my drivers, including GPU are all up-to-date as is Windows.

I have followed and tried all the hints and tips I can find on this issue!

I use a NVIDIA Geforce RTX 2080 Super GPU and all my OpenGL is up-to-date. I have tried changing the 3D settings around in NVIDIA Control Panel.

In fact, other than reinstalling Windows, I have tried everything I can think of.

I would love to use this software and make regular financial contributions as I have many projects in mind. At the moment I getting very frustrated.

Please help.

Best regards TheCottonMan

omichel, 2021-03-24:
Did you try to start Webots using the [safe mode](https://cyberbotics.com/doc/guide/starting-webots#safe-mode)?

TheCottonMan, 2021-03-24:
Hi omichel

Yes, I have, I have literally tried everything I can think of other than re-installing Windows which I am not prepared to do.

omichel, 2021-03-24:
It could be that a third party software has installed some conflicting DLL into your system folder, which conflicts with Webots.
Tracking this down is not easy, but can achieved with [dependency walker](https://www.dependencywalker.com) or similar tools.

Alternatively, you may try to install [Webots from the source](https://github.com/cyberbotics/webots/wiki) and start Webots from the MSYS2 console, hoping to get a useful error message.

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel