cyberbotics/webots #2914: WEBOTS crashes everytime

Fast-and-curious4, 2021-03-27:
**Describe the Bug**
I am unable to use the webots software as everytime I run it ,the software crashes causing the system to hang and restart.

**Expected behavior**
I expected the software to work fine because my system has the required hardware.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce RTX 150 2 GB

**Additional context**
Add any other context about the problem here.

ad-daniel, 2021-03-29:
Hi, does it start without problems in [safe mode](https://cyberbotics.com/doc/guide/starting-webots#safe-mode)?

Fast-and-curious4, 2021-03-29:
Yes, it works perfectly in safe mode but I just wanted to know the reason
because many of my friends did not have to do anything like safe mode.

ad-daniel, 2021-03-29:
It could be that the world being loaded by default or the controllers have problems and is causing it to hang/freeze. You should try loading a different world: `webots /path/to/different/world` from command line (one of the sample ones for example) and if that works too then the issue is with the world/controllers you're creating (infinite loops for example). Loading webots in pause mode `webots --mode=pause` can also help, it loads the world without actually running the simulation.

Fast-and-curious4, 2021-03-29:
Yes,the world is being loaded by default and it automatically starts
running the simulation.Thnx for your help.

ad-daniel, 2021-03-29:
You can change that behavior in the preferences: `tools > preferences > startup mode`.
Is the issue solved then?

Fast-and-curious4, 2021-03-29:
I have already created environment variable for safe mode. The simulation
does not start automatically in safe mode but earlier it was the case. Even
then am I required to follow your steps?

ad-daniel, 2021-03-29:
You don't need to, safe mode can be used to find out whether the issue is your world/controller or if it's a OpenGL related problem. After sorting that out, you should unset the environmental variable and use Webots without safe mode.

What I meant is if in the future you still wish to launch Webots without it automatically running the simulation on startup, then you can change the preferences or use the `--mode=pause` as mentioned above to prevent that.

Fast-and-curious4, 2021-03-29:
Thnx for your help. I do not have much knowledge about environment
variables and related stuff.

ad-daniel, 2021-05-08:
I'm closing this for now. If you still need help feel free to re-open it.

xishengping, 2023-08-21:
My problem is that when I click on a node in the scene tree, webots will freeze and not respond. I set --mode=pause, and when it is turned on in safe mode, webots will crash.
[image]