cyberbotics/webots #2920: camera detection bug with Humanoids

tarithj, 2021-03-29:
**Describe the Bug**
camera image recognition picks up humanoids that are not in view and returns invalid data

**Steps to Reproduce**
1. Download the world
[a.tar.gz](https://github.com/cyberbotics/webots/files/6219749/a.tar.gz)
2. place a simple robot with a camera
3. turn the robot while simulation is running
4. See bug

**Video**
https://user-images.githubusercontent.com/29999281/112796380-86b62200-9087-11eb-8830-5bcb12008682.mp4

**System**
 - Operating System: Ubuntu 20.10
 - Graphics Card: Nvidia Geforce GT710

stefaniapedrazzi, 2021-03-29:
The issue should come from the intermediate Solid node that has the scale set to [0.12, 0.12, 0.12].
This scaling factor is not handled correctly when computing the position of the detected object.

stefaniapedrazzi, 2021-03-29:
This issue should be fixed in #2921.
The fix will be included in the nightly build of Webots R2021a revision 1 available tomorrow (nightly 29-03-2021).

PR #2921 (2021-03-29): Fix camera object recognition detection using bounding sphere
Fix #2920: fix detected object size when using the bounding sphere that was not handling the scale.

Tasks:
* [x] fix computation of detected object size based on bounding sphere radius
* [x] port bounding sphere visualization to WREN
closed 2021-03-29 by stefaniapedrazzi