cyberbotics/webots #2931: CameraRecognitionObject.get_colors returns invalid data

tarithj, 2021-04-01:
**Describe the Bug**
CameraRecognitionObject.get_colors returns invalid data returns invalid data when it is called from inside a method in a class
example:
```python

class Side(Enum):
    left = "left"
    right = "right"

class FoundObjectInfo():
    def __init__(self, recognition_object: CameraRecognitionObject):
        self.recognition_object= recognition_object

    def get_color(self) -> ObjectColor:
        """A high level function that returns the color of the object"""
        colors = self.recognition_object.get_colors()
        print("this is what get_color returns when called inside a method", colors)
        if colors[0] > 1:
            print("found R")
            return ObjectColor.R
        elif colors[1] > 1:
            print("found G")
            return ObjectColor.G
        elif colors[2] > 1:
            print("found B")
            return ObjectColor.B        
```
when the ``FoundObjectInfo.get_color`` method is called it returns completely different data compared to using the ``CameraRecognitionObject.get_colors`` method after it is returned by the ``camera.getRecognitionObjects`` method

**Screenshots**
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card:  AMD Radeon (TM) R9 360

stefaniapedrazzi, 2021-04-01:
I just tested on the sample `projects/samples/devices/worlds/camera_recognition.wbt` and the color returned by `CameraRecognitionObject.get_colors` is correct.

> when the FoundObjectInfo.get_color method is called it returns completely different data compared to using the CameraRecognitionObject.get_colors method after it is returned by the camera.getRecognitionObjects method

Do you mean that `CameraRecognitionObject.get_colors` works correctly but not your `FoundObjectInfo.get_color` function?
If this is the case, then the problem is in your controller and not in the Webots API.
I would suggest you to try to make a deep copy of the `CameraRecognitionObject` object instead of shallow copy it when creating your `FoundObjectInfo`instance.

If you get the same wrong color calling directly `CameraRecognitionObject.get_colors` when retrieving it, then please also provide a simple example where this issue is reproducible.

tarithj, 2021-04-02:
[exp-car.zip](https://github.com/cyberbotics/webots/files/6250648/exp-car.zip)
I'm storing the ``CameraRecognitionObject`` in a list and when I call them I get
```
color:  [8.691694759885e-311, 1.1323610030567e-311, 1.0]
color:  [2.172923689984e-311, 1.1323609968275e-311, 0.0]
color:  [8.691694759885e-311, 1.1323610030567e-311, 1.0]
```
but due to the fact that no object in the world has that color and their are only basic colors (for example the SANATIZER robot's color is set to ``[0, 0 ,1]``). I think this is a bug.

tarithj, 2021-04-03:
when I move the object closer to the camera it detects the actual color

stefaniapedrazzi, 2021-04-07:
I quickly checked your world and if I print the CameraRecognitionObject color immediately when returned from the `camera.getRecognitionObjects()` the result is correct so the value returned by the Webots API.

As I already wrote you in the previous message, you should make sure to **deep copy** the `CameraRecognitionObject` when you store it in your list so that you don't refer to memory that could be changed by the Webots Python API during the simulation run.
For example you could create a dictionary out of the `CameraRecognitionObject` that can be used later without memory issues:
```
objects = camera.getRecognitionObjects()
copiedObjectList = []
for object in objects:
    copiedObjectList.append({'color': object.get_colors()})
all_object.append([copiedObjectList, components.Side.right])
```

> when I move the object closer to the camera it detects the actual color

This works just by chance because when you refer later to the `CameraRecognitionObject` objects stored previously the close object is still visible and thus still in the internal Webots Python API list of recognition objects.

closed 2021-04-07 by stefaniapedrazzi