cyberbotics/webots #2946: Question

zhinaebr, 2021-04-07:
**Describe the Bug**
When I run Webots it opens for only 3 seconds and then it closes without any errors or warning or even any reason.
We have gone and seen the others problems and we have added a value in environment variables that names WEBOTS_SAFE_MODE with a value of true but actually it didn't work.

**Steps to Reproduce**
1. Install Webots 
2. It opens
3. It closes after 3 seconds

**Expected behavior**
We were looking for the app to start but it didn't and it closed.

**Screenshots**

**System**
 - Operating System: [ Windows 10]
 - Graphics Card: [Intel(R) HD Graphics 3000]

**Additional context**

omichel, 2021-04-13:
Which version of Webots are you using?

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel