cyberbotics/webots #2948: Webots urdf/proto resource

anastasiapan, 2021-04-07:
Hello
I recently saw a related post on ROS Discourse (https://discourse.ros.org/t/urdf-with-webots/19665) that you plan to migrate from PROTO to urdf. My question is about the features only available in PROTO. The reason I post here and not on Discourse is because I didn't want to derail the conversation there.

So the problem is that we use the [connector](https://cyberbotics.com/doc/reference/connector) and to make that work with urdf we found a solution but it is a bit "hacky" and it uses a gazebo feature:

```xml
<gazebo>
    <joint name="rod_joint" type="fixed">
      <pose>0.285 0 0 0 0 0</pose>
      <parent>base_front_rod</parent>
      <child>base_back_rod</child>
    </joint>
  </gazebo>

  <!-- GAZEBO control -->
  <gazebo>
    <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
      <robotNamespace>$(arg robot_namespace)</robotNamespace>
    </plugin>
  </gazebo>
```
Would we be able to get our model in urdf? I am asking now because I am trying to be a bit proactive and see if PROTO file utilities will be gone. I can provide with both urdf or proto if you want.

lukicdarkoo, 2021-04-07:
We haven't decided on the architecture yet, but we plan to add a `<webots>` tag and put everything Webots specific inside:
https://github.com/cyberbotics/urdf2webots/issues/112

That means that Connectors, as well as other Webots specific devices, will be preserved.

We don't have plans on dropping the PROTO support, it will still be the native format.

anastasiapan, 2021-04-08:
Thank you very much for your response! I just misunderstood the post on Discourse then.

closed 2021-04-08 by anastasiapan