cyberbotics/webots #2952: Problem with and old NVIDIA geforce 635m device

Behrouz-m, 2021-04-09:
**Describe the Bug**
error on running the app. during loading a sample scene it crashes

**Steps to Reproduce**
when opening a sample

**Expected behavior**
opening the sample

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce 635m, **latest driver belongs to 2018**

**Additional context**
it might be the driver problem, but I had no problem with other 3d apps so far.

**windows log:**
```
The description for Event ID 1 from source NVIDIA OpenGL Driver cannot be found. Either the component that raises this event is not installed on your local computer or the installation is corrupted. You can install or repair the component on the local computer.

If the event originated on another computer, the display information had to be saved with the event.

The following information was included with the event: 

A TDR has been detected.
The application must close.

Error code: 7
 (pid=7120 tid=17952 webots-bin.exe 64bit)

Visit http://nvidia.custhelp.com/app/answers/detail/a_id/3633 for more information.

The message resource is present but the message was not found in the message table
```
second log:
```
Faulting application name: webots-bin.exe, version: 0.0.0.0, time stamp: 0x5fd9edfd
Faulting module name: nvoglv64.dll, version: 23.21.13.9135, time stamp: 0x5ab586ed
Exception code: 0xc0000409
Fault offset: 0x0000000000fe03a9
Faulting process id: 0x1bd0
Faulting application start time: 0x01d72d2c039bc87d
Faulting application path: C:\Program Files\Webots\msys64\mingw64\bin\webots-bin.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\nvlti.inf_amd64_83a389b28f4c421e\nvoglv64.dll
Report Id: 0def7aaa-48d6-40a5-b56e-9462c1f1b4f2
Faulting package full name: 
Faulting package-relative application ID: 
```

Behrouz-m, 2021-04-09:
I checked other issues, other users have the same problem. please check the log that I posted above.

omichel, 2021-04-09:
This is a NVIDIA bug. We cannot do much to fix it. If you cannot upgrade your graphics driver, try to downgrade it or upgrade your graphics card to a more recent one (this one was released 10 years ago).

Behrouz-m, 2021-04-09:
I do not think that is correct. all other software and even games work fine. only this software has this problem.
if you need more information/reports on this I would be happy to help.
I checked other issues, other users have the same problem. please check the log that I posted above.

Behrouz-m, 2021-04-09:
I uninstalled Nvidia Nsight before. I try to reinstall the driver. hope it fixs it.

Behrouz-m, 2021-04-09:
i reinstalled the driver. did not work.

Behrouz-m, 2021-04-16:
Well, I started Webots in safe mode. Then I opened another sample project. After closing and reopening in normal mode, it worked.
I think some of the samples have problems with Nvidia drivers. 
In my case following example had this problem:
[image]

closed 2021-11-15 by BenjaminDeleze