cyberbotics/webots #2959: Utilize types in the Python docs

lukicdarkoo, 2021-04-12:
Maybe, the Python documentation would be more readable by utilizing the Python types. For example, instead of:
```python
from controller import Node

class Node:
    def getId(self):
    def getDef(self):
    def getParentNode(self):
    def isProto(self):
    def getFromProtoDef(self, name):
```
we could write
```python
from controller import Node

class Node:
    def getId(self) -> int:
    def getDef(self) -> str:
    def getParentNode(self) -> Node:
    def isProto(self) -> bool:
    def getFromProtoDef(self, name : str):
```

The same applies to the implementation.

omichel, 2021-04-12:
Since which version of Python were these python types introduced?

lukicdarkoo, 2021-04-12:
From 3.5:
https://docs.python.org/3.5/library/typing.html

I am not whether it is backported to 2.7, but I guess we will drop the support for 2.7 soon

omichel, 2021-04-12:
Yes, we will drop 2.7 soon, so, it shouldn't be an issue. I just wanted to check this feature was not too recent (e.g., Python 3.8 or the like).

ygoumaz, 2022-11-08:
Closing as duplicate of #2385.