cyberbotics/webots #2963: Utility of collectHiddenKinematicParameters function

ad-daniel, 2021-04-12:
The way the function [collectHiddenKinematicParameters](https://github.com/cyberbotics/webots/blob/180777b4f9114e336eb39f70f8feda42a7ffc552/src/webots/nodes/WbSolid.cpp#L2905-L2931) is implemented leads me to believe that the `BallJoint` case has not been considered (i.e position3). I've made some quick tests to check if by any chance this function was part of the saving of hidden parameters mid-execution but I couldn't really manage to break anything. The saving of the hidden fields for a BallJoint works fine.

So I'm not entirely sure whether this function (and the analogous on in `WbGroup`) serve any purpose. They don't appear to be called anywhere as far as I can tell, but if it's necessary to keep them, then the question becomes whether the `BallJoint` case should also be handled.

omichel, 2021-04-12:
We should definitely clean this up by removing this method if it is not used.

stefaniapedrazzi, 2021-04-12:
Just for note, the BallJoint parameter3 has been introduce after the `collectHiddenKinematicParameters` function has been implemented.

stefaniapedrazzi, 2021-04-12:
> So I'm not entirely sure whether this function (and the analogous on in WbGroup) serve any purpose. They don't appear to be called anywhere as far as I can tell

Please make sure that when testing the BallJoint node is an internal field of a PROTO and not visible in the scene tree because only in this case it is needed to save hidden fields.

ad-daniel, 2021-04-12:
I was wrong, it does appear to play a role in the `writeParameters` of `WbGroup` [here](https://github.com/cyberbotics/webots/blob/180777b4f9114e336eb39f70f8feda42a7ffc552/src/webots/nodes/WbGroup.cpp#L344). So it's more the latter case it seems, and could likely lead to a bug.

> Please make sure that when testing the BallJoint node is an internal field of a PROTO and not visible in the scene tree because only in this case it is needed to save hidden fields.

 Ok, I'll look into it more

closed 2021-04-14 by ad-daniel