cyberbotics/webots #2965: Cannot package the feature-robocup-controllers branch: Missing file: projects/samples/contests/robocup/protos/icons/Darwin-opHinge2.png

schra, 2021-04-12:
**Context**
I'm building a custom Docker image for the RoboCup. For this I wanted to create a package of the `feature-robocup-controllers` branch. Since my host system is Arch Linux and not Ubuntu, I can't directly package Webots on my host system. Instead, I tried to do this in a Ubuntu Docker container.

**Describe the Bug**
I cannot package the feature-robocup-controllers branch because of a missing file.

**Steps to Reproduce**

Latest Git commit at branch `feature-robocup-controllers` as of right now:

    git checkout 107c5b75632ac9cb133fc4e0417b43e04190153d

Run Ubuntu Docker container:

    docker run -it --rm -v "$PWD"/webots:/webots ubuntu:focal

Inside the Docker container I tried to package Webots:

    DEBIAN_FRONTEND=noninteractive ./scripts/install/linux_compilation_dependencies.sh
    DEBIAN_FRONTEND=noninteractive ./scripts/install/linux_optional_compilation_dependencies.sh

    # libxslt1.1 is a missing dependency 
    apt-get install libxslt1.1
    source /opt/ros/noetic/setup.bash
    update-alternatives --install /usr/local/bin/python python /usr/bin/python3.8 3
    cd /webots
    make distrib -j10

Error message:

```
[...]
# packaging
#
# removing files and folders not in git repository
# generating project files
missing icon: projects/samples/contests/robocup/protos/icons/Darwin-opHinge2.png
# updating proto cache
Warning: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
# checking submodules update
# Submodule 'resources/webots_ros':
# - 'Current' and 'origin/master' commits IDs mismatch.
#   - 'Current' commit id: fb78c3ba4696fd2b34cfa2a39d8a4a9bd193ae16
#   - 'origin/master' commit id: 4a52aa2b89955ae73dcc762e59c8b5e5fb0eca03
# Submodule 'src/glm':
# - 'Current' and 'origin/master' commits IDs mismatch.
#   - 'Current' commit id: dd48b56e44d699a022c69155c8672caacafd9e8a
#   - 'origin/master' commit id: 761a842a595f58acfc01cc5c7760ff351feaaece
# Submodule 'src/stb':
# - 'Current' and 'origin/master' commits IDs mismatch.
#   - 'Current' commit id: f67165c2bb2af3060ecae7d20d6f731173485ad0
#   - 'origin/master' commit id: c9064e317699d2e495f36ba4f9ac037e88ee371a
# building distribution source
# creating webots.deb version R2021bMissing file: projects/samples/contests/robocup/protos/icons/Darwin-opHinge2.png
make[1]: *** [Makefile:80: all] Error 255
make: *** [Makefile:62: distrib] Error 2
```

Locally I only have a small change that shouldn't cause this problem (I made a fix that I can package Webots inside a Docker container at all):

```
$ git status
On branch feature-robocup-controllers
Your branch is up to date with 'origin/feature-robocup-controllers'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   scripts/packaging/Makefile

no changes added to commit (use "git add" and/or "git commit -a")

$ git diff | cat
diff --git a/scripts/packaging/Makefile b/scripts/packaging/Makefile
index 8692bc12d..6e7588d9f 100644
--- a/scripts/packaging/Makefile
+++ b/scripts/packaging/Makefile
@@ -73,7 +73,7 @@ all: $(WEBOTS_DISTRO_EXE) $(ISCC_FORMATTER) $(XVFB)
 	+@echo "# generating project files"
 	+@./generate_projects_files.py > files_projects.txt
 	+@echo "# updating proto cache"
-	+@$(WEBOTS) --update-proto-cache
+	+@xvfb-run $(WEBOTS) --update-proto-cache
 ifeq ($(OSTYPE),darwin)
 	+@echo "# checking RPATH system"
 	+@./check_rpath.py

$ git show --name-only | cat
commit 107c5b75632ac9cb133fc4e0417b43e04190153d
Author: Olivier Michel <Olivier.Michel@cyberbotics.com>
Date:   Mon Apr 12 16:24:33 2021 +0200

    Handle multiple fallen robots

projects/samples/contests/robocup/controllers/referee/referee.py
```

Note that I don't have this problem with 9a20486bf5b8404d100ab00e0e46149dc04915f8. There I could package everything as expected.

Update: A new commit was just pushed and with 501b814aa5f168112a98af67e27ed2ac4f07fa92 I have the same problem.

/cc @medrimonia @MaikePaetzel

omichel, 2021-04-12:
We introduced a new proto file, but forgot to include a corresponding icon. This is now fixed in https://github.com/cyberbotics/webots/pull/2888/commits/7d51c5258494c78976ce185fbc772be04a40f692. Can you try again?

schra, 2021-04-12:
Thank you for the speedy response :) I can now package everything as expected.

closed 2021-04-12 by schra