cyberbotics/webots #2966: Streaming server instructions

anastasiapan, 2021-04-13:
**Describe the Bug**
I am trying to start a streaming server but Webots display on the browser hangs. I am not sure if this is a bug or maybe a documentation issue?

**Steps to Reproduce**
***With the default example:***
1. Follow the instructions: https://cyberbotics.com/doc/guide/web-simulation#quick-start
2. Open the interface on browser:
```
file:///usr/local/webots/resources/web/streaming_viewer/index.html
```
3. In the `Connect to` field type:
```
ws://localhost:1999/session?url=webots://github.com/cyberbotics/webots/branch/develop/projects/languages/python/worlds/example.wbt
```
4. Webots just hangs in the "Starting simulation" message
5. I can see that the session and simulation servers start:
```
Started default session server (9628)
Started default simulation server (9629)
```
and that I opened the ports:
```
python3   10055            root    7u  IPv6  97272      0t0  TCP *:1999 (LISTEN)
python3   10055            root    8u  IPv4  97273      0t0  TCP *:1999 (LISTEN)
python3   10056            root    4u  IPv4  94958      0t0  TCP *:2000 (LISTEN)
python3   10056            root    5u  IPv6  94959      0t0  TCP *:2000 (LISTEN)
```
***With my project:***
I also tried to start the servers with my own configuration:
1. simulation/simulation_config.json:
```
{
    "port": 2001,
    "projectsDir": "/path/to/my/own/webots/project"
}
```
2. session/session_config.json:
```
{
    "port": 2002,
    "server": "localhost",
    "simulationServers": [
      "localhost:2001"
    ]
}
```
3. server.sh:
Everything the same except lines 43-44:
```
SESSION_CONFIG="config/session/session_config.json"
SIMULATION_CONFIG="config/simulation/simulation_config.json"
```
Again the same thing I can see the ports are open and when I type `ws://localhost:2002` in the `Connect to` field the interface just hangs again at the same place. For this configuration though it might be something I miss but it looks like it is the same behavior.

**Expected behavior**
I would like to see the simulated environment on my browser - both for the example and my own world.

**System**
 - Operating System: Ubuntu 20.04
 - Graphics Card: Nvidia Geforce RTX 2060

stefaniapedrazzi, 2021-04-13:
Related to #2891

anastasiapan, 2021-04-13:
> Related to #2891

Thanks I will keep an eye to that issue until it's closed!

closed 2021-11-15 by BenjaminDeleze