cyberbotics/webots #2981: Encountered "no available '<extern>' robot controller found" when using webots_ros2

BruceXSK, 2021-04-18:
I opened the epuck_world.wbt in webots and confirm the controller is <extern>. Then I run my own ROS2 python code which is:
```Python3
import rclpy
from webots_ros2_core.webots_node import WebotsNode

class ScoutDriver(WebotsNode):

    def __init__(self, args):
        super().__init__('scout_driver', args=args)

def main(args=None):
    rclpy.init(args=args)
    scout_driver = ScoutDriver(args=args)
    scout_driver.destroy_node()
    rclpy.shutdown()
```

The first time I run this code (`ros2 run scout_webots driver`), everything was fine.
Console showed that
```
INFO: Starting extern controller for robot "e-puck".
INFO: '<extern>' controller exited successfully.
```
Because there was no `rclpy.spin()` in my code, so the code will exit immediately.

Then I re-run the `driver` code, webots gave a warning like:
[image]

Then as long as I run the code, I will get this warn. Unless click `Reset Simulation` or `Reload World` in webots to reset the whole enviroment.

Thanks a lot

BruceXSK, 2021-04-18:
Well, I solved this problem just now. The given example code of webots_ros2 is both launching the driver and webots. But I want to run them separately so that I can debug the code without re-run the simulation environment. But I just run the webots by clicking the icon in Applications in ubuntu. So maybe there are some environment variables added by .bashrc missing in webots.
So start the webots in terminal can solve this problem.
I don't know the exact reason, but it works now anyway.

closed 2021-04-18 by BruceXSK

omichel, 2021-04-19:
It could be that the `WEBOTS_HOME` environment variable is missing when you start Webots from the icon.

BruceXSK, 2021-04-20:
> It could be that the `WEBOTS_HOME` environment variable is missing when you start Webots from the icon.

Yeah I think so. Thanks for your reply