cyberbotics/webots #2982: Webots crashes after changing Lidar-type

KNTRL9, 2021-04-19:
**Describe the Bug**
Webots R2021a crashes consistently after changing the "type" of the Lidar-Node from "rotating" to "fixed" or vice versa and pressing "Reset Simulation".

**Steps to Reproduce**
1. Go to 'Webots\projects\samples\devices\worlds\lidar.wbt'
2. Change the type of the Lidar-Node to "fixed"
3. Reset the simulation
4. Webots Crashes either immediately or after a short freeze

Out of 10 times, it crashed 8 times. The other 2 times it took several more seconds where I could Play/Pause/Reset until it crashed.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Windows 10 (Same for Ubuntu 18.04.5)
 - Nvidia Quadro K2100M on Windows Device

**Additional context**
Add any other context about the problem here.

stefaniapedrazzi, 2021-04-19:
After changing the `Lidar.type` value, the internal image size should be updated but it is not the case.
So when resetting the internal image the wrong size is used causing the crash due to write outside the boundaries.

I'm also inspecting the other issues caused by the missing update in order to find the better fix.

omichel, 2021-04-20:
@stefaniapedrazzi: can we close this issue?

stefaniapedrazzi, 2021-04-21:
This issue is fixed in #2983.
Given that the dimensions of the returned data depends on the `Lidar.type`, we had to remove the possibility to change it during simulation run and force saving and reverting the simulation in order to apply the type change.

closed 2021-04-21 by stefaniapedrazzi