cyberbotics/webots #2986: Animation json file gets deleted (when invoked by supervisor)

Adman, 2021-04-19:
**Describe the Bug**
When I create a HTML animation from within the Supervisor's (python) code, the json file gets deleted after Webots GUI is closed. If I create the animation by clicking on the icon in the GUI and close Webots GUI, the json file stays present on my disk, which is OK.

I checked the docs but couldn't find anything relevant that would give me a hint how to fix this problem. 

**Steps to Reproduce**

```python
supervisor.animationStartRecording("filename.html")
....
supervisor.animationStopRecording()
supervisor.simulationQuit(0)
```

**Expected behavior**
The json file should **not** be deleted.

**System**
 - Operating System: Linux Ubuntu 20.04

omichel, 2021-04-20:
> # I also tried calling time.sleep() here, but that did not help

Instead did you try `supervisor.step(time_step)` or `supervisor.step(100 * time_step)`?

Adman, 2021-04-20:
Tried it now, but doesn't work either. I also call `supervisor.simulationSetMode(supervisor.SIMULATION_MODE_PAUSE)` after `step` but the same problem occurs.

stefaniapedrazzi, 2021-04-21:
> Also, I checked the resulting json and it looks like it is missing the header -- It contains only lines with {"time" ...}. If I run it from GUI, it is correct, like the example below

It is correct that the header is not in the file while recording the animation because it is prepended when finalizing it, i.e. after the `supervisor.animationStopRecording()` is called.

Adman, 2021-04-21:
@stefaniapedrazzi Can you reproduce the issue or is this just a problem on my side? Is it possible that the `animationStopRecording` is doing something wrong while prepending headers and somehow deletes the file?

stefaniapedrazzi, 2021-04-21:
No, I cannot reproduce this issue.
The only case in which the JSON file is removed is if the world is closing while the animation is finalized and this could happen when calling the `supervisor.simulationQuit()` too early.

One first thing you should try is to remove the `supervisor.simulationQuit(0)` and manually quit Webots from the UI.
Then you should try to add many simulation steps as suggested by @omichel before quitting the simulation.
Note that it is useless to call `supervisor.simulationSetMode()`.

Adman, 2021-04-21:
Thanks. The problem was actually in my code, which resulted that stop recording was not called.

After all, it is not necessary to call `supervisor.step`, the recording is saved completely.

Sorry for the raising this and thanks for your support :)

closed 2021-04-21 by Adman