cyberbotics/webots #2996: Fail to load generic window library (MATLAB)

mael25, 2021-04-22:
**Describe the Bug**
The MATLAB controllers crash in the develop branch, because they fail to load the generic window library (see Screenshots).

**Steps to Reproduce**
1. Open and run `nao_matlab.wbt `or `youbot_matlab.wbt`
2. See error

**Expected behavior**
The library loads correctly and the controller does not crash.

**Screenshots**
[image]

**System**
 - Operating System: Linux Ubuntu 20.04.2 LTS
 - Graphics Card: NVIDIA GeForce GTX 1650 Mobile / Max-Q

**Additional context**
I think the issue might be due to the fact that the files of the new robot_window in `include/controller/c/webots/plugins/robot_window` are not included in `lib/controller/matlab/allincludes.h` and not mentioned in `lib/controller/matlab/launcher.m`.

mael25, 2021-04-27:
Hello @stefaniapedrazzi, so I have checked in more details where the error happens:

1. `lib/controller/matlab/launcher.m` (l.150) 
- calllib('libController', 'wb_robot_init');
2. `projects/robots/kuka/youbot/controllers/youbot_matlab/youbot_matlab.m` (l.14)
- passive_wait(4.0)
3. `projects/robots/kuka/youbot/controllers/youbot_matlab/passive_wait.m` (l.4)
- if (wb_robot_step(32)==-1)
4. `resources/projects/plugins/robot_windows/generic/generic.c` (l.16): 
- if(!wbu_generic_robot_window_handle_messages(message)) 
5. At this place, the error (see above) appears.

So basically, the error occurs the first time `wb_robot_step()`  is called inside the Matlab controller. Then, it looks in `libgeneric.so ` (generated from `resources/projects/plugins/robot_windows/generic/generic.c`) for `wb_robot_window_step()`. Finally, inside `wb_robot_window_step()`, it fails to find `libgeneric_robot_window.so` (generated from `resources/projects/libraries/generic_robot_window/generic.c`) necessary to use `wbu_generic_robot_window_handle_messages()` and the other generic window functions.

I guess it might be because it looks inside the wrong place for `libgeneric_robot_window.so`, but I have trouble to find where/how to define the search location of the library... Do you have any idea ?
Thanks!

stefaniapedrazzi, 2021-04-28:
Something to check could be to hardcode the path to `libgeneric_robot_window.so` using RPATH, by adding the following lines in the `resources/projects/plugins/robot_windows/generic/Makefile` (just after ``LIBRARIES = -lgeneric_robot_window``).
```
ifeq ($(OSTYPE),linux)
 LIBRARIES += -Wl,-rpath,$(WEBOTS_HOME_PATH)/lib/controller
endif
```

mael25, 2021-04-29:
Thanks, it has solved the problem!

closed 2021-05-17 by stefaniapedrazzi