cyberbotics/webots #3000: Allows external controller to reconnect after been killed

ShuffleWire, 2021-04-22:
Hello, 
First of all, I wish to say that Webot is so nice, thank a lot for it !
Moreover I wish to contribute, but I'm new to it, and the solution I've found could be pushed with a regular PR, but my brain melt when I've tried to do so. Please help if possible :)

**Describe the Bug**
I'm working with external C controller, which work very well.
However, if the controller died, it can't call the cleanup function, and when I restart it, I get a error message from webots because he think that the controller is still running, so he can't find a spot for the new incoming connection.
I'm often in this configuration, because I'm debugging with gdb, and we kill everything when I've found the location of a bug, for instance.

**Workaround**
The workaround is to restart the simulation via the GUI each time, but it's not very handy...

**Solution**
I wish the controller could reconnect automatically.
I've tested a solution, which is to add the following code in `src/webots/control/WbControlledWorld.cpp`, between lines : 222 and 223
It simply remove the dead controller connection as usual, then recreate a new one. (a bit of code duplication from the line above, my bad... It might be possible to avoid that).
The trick is to use the name of the robot to know which dead controller should be kill. (which is fine for me). Otherwise, if no name is provided, the code execute as usual (scanning for free spot).
Maybe a more elegant solution is possible, by reassigning a socket id, for instance, but I'm first trying to get it working. I've noticed the robot_id, which maybe could be used for that, but I've not investigated in that direction.
It's have not been tested in every way possible, but I'm not seeing major issues coming.
The main thing to check would be leak of memory, unclosed socket and so on, because I'm not sure how is it performed.

```c
     robotName = buffer;
      delete[] buffer;
    }
    foreach (WbRobot *const robot, robots()) {
+      if (robotName == robot->name() && robot->controllerName() == "<extern>" && robot->isControllerStarted()) {
+        updateRobotController(robot);
+        WbLog::info(tr("Restarting extern controller connection for robot \"%1\".").arg(robot->name()));
+        mRobotsWaitingExternController.append(robot);
+        startControllerFromSocket(robot, socket);
+        return;
+      }
      if (robot->isControllerStarted())
        continue;
```
I would be glad if I got feedback about it :) And if this is a correct solution, I would be happy to get some help to push it properly, because I would enjoy contribute, but a need to get the whole PR process working!

Thank in advance :)

omichel, 2021-04-23:
Your patch looks pretty good to me. Feel free to open a new PR to propose it. I just invited you as an official member of the Committers group for our repos, so that you don't even need to fork the repo to propose a new PR.

ShuffleWire, 2021-04-23:
Thanks for you return.
However, I've trouble pushing code, that what I've done : 

git branch fix-reconnect-failed-controller
git checkout fix-reconnect-failed-controller
git push --set-upstream origin fix-reconnect-failed-controller

and I got that : 

Username for 'https://github.com': ShuffleWire
Password for 'https://ShuffleWire@github.com': <wrote my password here>
remote: Permission to cyberbotics/webots.git denied to ShuffleWire.
fatal: unable to access 'https://github.com/cyberbotics/webots.git/': The requested URL returned error: 403

I'm confuse on the way to go, I should be missing something...  I've read about the whole fork/PR process, but apparently you grant me right to use directly my own branch on the webots repo. I'm correct ?

"I just invited you as an official member of the Committers group for our repos" I've not notice anything particular about it, should I make any specific action ?

PS : not sure if discussing it here is the best place, feel free to move the discussion elsewhere is needed.

stefaniapedrazzi, 2021-06-17:
Fixed in #3189.

PR #3189 (2021-06-17): Fix reconnecting extern controllers
Replaces #3005.
Fixes: #3000 and #1903

> If the controller died, it can't call the cleanup function, and when I restart it, I get a error message from webots because he think that the controller is still running, so he can't find a spot for the new incoming connection.

> It simply remove the dead controller connection as usual, then recreate a new one.
The trick is to use the name of the robot to know which dead controller should be kill.
Otherwise, if no name is provided, the code execute as usual (scanning for free spot).

In case a single extern control [... cut]
closed 2021-06-17 by stefaniapedrazzi