cyberbotics/webots #3007: Can't download your executable program

zengyili, 2021-04-24:
After a period of normal downloading, the download will suddenly fail, and it will show that the server has rejected the user's download request

I am from China and I use a VPN to access GitHub. Will this affect the download process? I have searched a lot of websites but can't solve my problem

please!!help me!!

ad-daniel, 2021-04-26:
It does occasionally happen (to me happened a lot especially when downloading older versions) but it's an issue on Github's side. What you can try is using a browser add-on like DownThemAll (it's open source) which usually have features to resume a failed download. Some browsers also do, but more often than not they fail at it.

zengyili, 2021-04-26:
@ad-daniel 
Thank you very much for your solution. I am a novice. This is my first time asking questions on GitHub. The enthusiasm of the community warms my heart.

After I asked the question, I tried to solve the problem by myself. I can download the source code normally, so with the help of Google, I tried various ways to solve a bunch of problems, and finally managed to compile the executable successfully. Program, I am a complete novice, this is the first open source project I compiled。

Thank again for you generous help and valuable advice or tip.it certainlywill help me a lot in the future.
 

After all,I love Google,It is called an accelerator of innovation and it is veritable

closed 2021-04-27 by lukicdarkoo