cyberbotics/webots #3008: Cannot Install Webots on Ubuntu 20.04 with ARM

LucasBruzzone, 2021-04-25:
**Describe the Bug**
I have a Macbook Pro with M1 chip. I'm using UTM to have an Ubuntu 20.04 Virtual Machine with ARM (https://mac.getutm.app/gallery/ubuntu-20-04).

None of the installation methods work, I've tried them all a couple of times because it's not supported.

**Steps to Reproduce**
1. Go to https://cyberbotics.com/doc/guide/installation-procedure#installation-on-linux
2. Try every Linux installation there is.
3. See that each one fails.

**Expected behavior**
Webots should get installed

**System**
 - Operating System: [Linux Ubuntu 18.04]

**Additional context**

Thanks for this great product, this is just a minor issue with the new chip.
Let me know if you need anything from me to help

stefaniapedrazzi, 2021-04-26:
Could you please also add the error messages you get trying to install Webots with the different packages.

Do you have issues with Ubuntu 20.04 (title) or 18.04 (description)?

lukicdarkoo, 2021-04-27:
@LucasBruzzone Which architecture did you chose for the virtual machine? Note that Webots packages are built for x86/x64 and they don't work on ARM.

What is the output of the `uname -a` command (from the virtual machine)?

BenjaminDeleze, 2021-11-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-11-15 by BenjaminDeleze

SarvagyaVaish, 2024-02-07:
Is webots still not supported on ARM? I was hoping to run it on a jetson nano.

omichel, 2024-02-07:
Although Cyberbotics do not officially supports Webots on ARM, it should be possible for you to compile Webots from the source for ARM. You will probably need to modify the Makefiles, get the dependencies for ARM, etc.

calyle, 2025-08-22:
Does any one can provide `libpico` source code? I cannot find it online.