cyberbotics/webots #3017: Wrong angle mapping relationship between Webots and ROS

BruceXSK, 2021-04-28:
There is a VLP-16 lidar on my robot. To convert the point cloud data to ROS message conveniently, I rotate the radar 90 degrees clockwise along the y-axis, so the correspondence of the coordinate axes is Front-Z, Left-X, Up-Y. And then I rotate 15 degrees along the x-axis. As the picture shows:
[image]
And the rotation of the Lidar:
[image]

I set the pitch angle of joint of laser_link with base_footprint in urdf file to 0.2618, aka, pi/12 or 15 degrees.
And revised the order of xyz of points in publish code:
```python
    def _lidar_publish(self):
        data = self.lidar.getPointCloud(data_type='buffer')
        if data:
            msg = PointCloud2()
            msg.header.stamp = rospy.Time.now()
            msg.header.frame_id = 'laser_link'
            msg.height = 1
            msg.width = self.lidar.getNumberOfPoints()
            msg.point_step = 20
            msg.row_step = 20 * msg.width
            msg.is_dense = False
            msg.fields = [
                PointField(name='y', offset=0, datatype=PointField.FLOAT32, count=1),
                PointField(name='z', offset=4, datatype=PointField.FLOAT32, count=1),
                PointField(name='x', offset=8, datatype=PointField.FLOAT32, count=1)
            ]
            msg.is_bigendian = False
            msg.data = data
            self._lidar_pub.publish(msg)
```
But in the rviz, the point cloud has a little angle difference with the ground:
[image]

Then I tried to regulate the pitch angle in urdf. When it is 0.3305 (about 19 degrees)
[image]
The points belong to the ground can overlap with the ground, but the pointcloud belong to the wall is not perpendicular to the ground. 

And I just tried, if the lidar is parallel to the horizontal plane, that is to say the pitch of the lidar is 0, the point cloud is correct in ROS.
So, are there any steps I did wrong? Thanks a lot

lukicdarkoo, 2021-04-28:
Please specify the ROS and Webots versions

BruceXSK, 2021-04-28:
> Please specify the ROS and Webots versions

Sorry
Ubuntu-20.04  ROS-noetic  Webots-R2021a

lukicdarkoo, 2021-04-28:
Why do you publish data in Python? If you set the `ros` node in your world then Webots should automatically create necessary topics:
https://cyberbotics.com/doc/guide/using-ros#using-the-standard-ros-controller

BruceXSK, 2021-04-29:
> Why do you publish data in Python? If you set the `ros` node in your world then Webots should automatically create necessary topics:
> https://cyberbotics.com/doc/guide/using-ros#using-the-standard-ros-controller

Well, because I need to change the order of xyz axes of data, so I mimic the method of publishing point clouds in webots_ros2.
And I also tried to use the standard ros controller according to your suggestion. But I also need to revise the code related to the coordinate of points in WEBOTS_HOME/projects/default/controllers/ros/RosLidar.cpp:
```c++
void RosLidar::publishPointCloud() {
  const WbLidarPoint *pointCloud = mLidar->getPointCloud();
  sensor_msgs::PointCloud cloud;
  cloud.header.stamp = ros::Time::now();
  cloud.header.frame_id = mRos->name() + '/' + RosDevice::fixedDeviceName();
  sensor_msgs::ChannelFloat32 layerChannel;
  layerChannel.name = "layer";
  for (int i = 0; i < mLidar->getNumberOfPoints(); i++) {
    geometry_msgs::Point32 point;
    /******** Revised *******/
    point.x = -pointCloud[i].x;
    point.y = pointCloud[i].z;
    point.z = pointCloud[i].y;
    /*************************/
    cloud.points.push_back(point);
    layerChannel.values.push_back(pointCloud[i].layer_id);
  }
  cloud.channels.push_back(layerChannel);
  mPointCloudPublisher.publish(cloud);
}
```
And rebuild the binary "ros"

The result is the same, the the wall is not perpendicular to the ground either:
[image]
[image]
(I mean the white points belong to wall is not perpendicular to the points belong to the ground, not the grid of rviz)

But if the lidar is set parallel to the horizontal plane, the point cloud shows normal
[image]
[image]

So I think the reason of tilted wall is not the method of reading data but the rotation of lidar.

By the way, the method of reading point cloud data in webots_ros2 which use buffer is much faster than the method in RosLidar.cpp which deepcopy the coordinate of every point.

lukicdarkoo, 2021-04-29:
Can you try Webots R2021b and this approach:
https://github.com/cyberbotics/webots_ros/blob/2bfc812411d82cf82427d0e414be3fc34a04e988/launch/pioneer3at.launch#L31

BruceXSK, 2021-05-05:
> Can you try Webots R2021b and this approach:
> https://github.com/cyberbotics/webots_ros/blob/2bfc812411d82cf82427d0e414be3fc34a04e988/launch/pioneer3at.launch#L31
> 
> You can try the `pioneer3at.launch` simulation, it performs mapping just fine.

Sorry for taking so long to reply to you, and happy Worker's Day by the way.

I tried the `static_transform_publisher` way referred to `pioneer3at` simulation, and got same result. So I thought the tf published by `static_transform_publisher` is same as the one published by `robot_state_publisher` by parsing urdf model.

Then I tried Webots R2021b, what suprised me was that I can see the lidar point cloud (CTRL+F8) in webots now, which was not visible in R2021a. And the point cloud shows:
[image]
So it shows the same as in rviz, and no matter how I change the pitch angle in urdf, the point cloud displayed in webots will not change. So I think the problem may not come from tf.

So do you have any more thoughts on this issue? Is there anything else I can try?
Thanks

lukicdarkoo, 2021-05-05:
Maybe it is related to this:
https://github.com/cyberbotics/webots/issues/37

BruceXSK, 2021-05-20:
> Maybe it is related to this:
> #37

Thanks, this issue inspired me.
As @DavidMansolino said
> Instead of tilting the sub-cameras, it seems safer to create a bigger field of view and use only the interesting part

But I have a real robot carry a VLP-16 lidar, and I have to tilt the lidar to get more vision front of the robot. So I have to reproduce this effect in the simulation environment.
But his statement still inspired me.

I made a lot of attempts. At first I set the `spherical` to false, but there are still some problems.
But when I tried to increase the `verticalFieldOfView` of lidar, I found that the error between lidar points and the real position became smaller.
The angle between adjacent layers of VLP-16 is 2°, so I set the `numberOfLayers` to 64 and set the `verticalFieldOfView` to 2.199(126°). I got the following result:
In webots:
[image]
In rviz:
[image]
The points fitted to the objects.
So what I need to do is to select the middle 16 layers as the final point cloud.

I guess the cause of this problem may be related to the sampling of the depth camera used inside the lidar.
Although this approach solves my problem temporarily, I still hope this bug can be fixed finally. After all, this would cause some coding troubles and would waste a lot of computing resources.

lukicdarkoo, 2021-05-20:
> I guess the cause of this problem may be related to the sampling of the depth camera used inside the lidar.
> Although this approach solves my problem temporarily, I still hope this bug can be fixed [...]

We should verify whether the problem comes from:
- The depth image. We can use the RangeFinder to verify that and there are a few responsible shaders in https://github.com/cyberbotics/webots/tree/master/resources/wren/shaders
- The way the point cloud is calculated:
https://github.com/cyberbotics/webots/blob/226d90905dc3f46860383153fadfe5f3a24ed8a2/src/webots/nodes/WbLidar.cpp#L351-L377

BruceXSK, 2021-05-20:
> We should verify whether the problem comes from:
>
> * The depth image. We can use the RangeFinder to verify that and there are a few responsible shaders in [...]

Thanks, I will close this issue.

closed 2021-05-20 by BruceXSK