cyberbotics/webots #3020: Bug with camera in feature-robocup-controllers branch

ar0usel, 2021-04-28:
During testing player controller from [feature-robocup-controllers](https://github.com/cyberbotics/webots/blob/feature-robocup-controllers) branch I meet several problems with camera.

Code for context:
https://github.com/cyberbotics/webots/blob/7751590bdf521edf75b3ca6295756dd6a04dc117/projects/samples/contests/robocup/controllers/player/player.cpp#L399-L408

**1.Getting image from webots**
As I understand from documentation, we can't directly access to `rgba_image` variable. For me helps:
```
for (int x = 0; x < width; x++)
  for (int y = 0; y < height; y++) {
    rgb_image[(x + y)*3 + 0] = camera->imageGetRed(rgba_image, width, x, y);
    rgb_image[(x + y)*3 + 1] = camera->imageGetGreen(rgba_image, width, x, y);
    rgb_image[(x + y)*3 + 2] = camera->imageGetBlue(rgba_image, width, x, y);
}
```

**2.Static for `rgb_image`**
With static variable  `rgb_image` after one successful cycle player controller crushes. This helps for me:
`unsigned char *rgb_image = new unsigned char[rgb_image_size];
`
**3.Empty image**
If i set `sensor_timestep` for camera equal not to 8, I always get empty image (`sensor_timestep` for all other sensors is set to 8). Now I do not manage to solve this issue.

P.S. If you agree with my thoughts about first two points, I can create pr and I will be very thankful if you help me with last.

omichel, 2021-04-29:
1. Yes, accessing directly the image data as it is done here should be equivalent to the solution you propose, but faster. Did you find out that it was not working? By the way, I believe your code sample is wrong, e.g., it should be like:
```c++
    rgb_image[(x + y * width) * 3 + 0] = camera->imageGetRed(rgba_image, width, x, y);
    rgb_image[(x + y * width) * 3 + 1] = camera->imageGetGreen(rgba_image, width, x, y);
    rgb_image[(x + y * width) * 3 + 2] = camera->imageGetBlue(rgba_image, width, x, y);
```
2. I agree, this is a bug and your patch seems correct, please go ahead to propose your patch.
3. I need to further investigate this one. I will get back here about it.

omichel, 2021-04-29:
Regarding 3.: you cannot use a value for the time step that is not a multiple of 8. Did you try with 16, 24, 32, 64, etc.?

ar0usel, 2021-04-29:
> Regarding 3.: you cannot use a value for the time step that is not a multiple of 8. Did you try with 16, 24, 32, 64, etc.?

Yes. My sensor time step is always multiplies of 8. 

> 1. Yes, accessing directly the image data as it is done here should be equivalent to the solution you propose, but faster. Did you find out that it was not working? By the way, I believe your code [...]

I checked the length of rgba image, and it is not equal to width*height*4. Firstly I thought that this causes player controller crash, but probably the main problem is static variable rgb_image. I will check it today!

omichel, 2021-04-29:
> I checked the length of rgba image, and it is not equal to width * height * 4.

How did you check this?

> Yes. My sensor time step is always multiplies of 8.

Did you try with 16? If so, do you get a blank image or no image at all?

ar0usel, 2021-04-29:
> How did you check this?

`strlen((char*)rgba_image)`

> Did you try with 16? If so, do you get a blank image or no image at all?

Blank image (`strlen((char*)rgba_image) = 0`) Moreover, if in the first message the camera time step is equal to 8, it shows one frame and then when I reset time step to 64 in the next message there is no new images.

ar0usel, 2021-04-29:
>`strlen((char*)rgba_image)`
I understood, that this check is no safe. 

> Did you try with 16? If so, do you get a blank image or no image at all?
I try with 16 and looks like that all sensors don`t return values, if time step is set not to 8.

omichel, 2021-04-29:
> >`strlen((char*)rgba_image)`

> I understood, that this check is no safe. 

Yes, it's not.

> I try with 16 and looks like that all sensors don`t return values, if time step is set not to 8.

It should. The following test:
```c++
              if (camera) {
                if (controller_time % camera->getSamplingPeriod())
                  continue;
```
should ensure that if the `controller_time` is 16, 32, 48, 64, etc. the result of the modulo (%) operation with the camera sampling (16) should be 0 and therefore it should not continue with the next sensor and proceed with sending sensor data. The `controller_time` is increased at 8 ms at each time step.

May I let you try to debug the problem by adding `printf` statements to check the behavior?

ar0usel, 2021-05-03:
Sorry for the long answer. Debugging shows that this *if* statement is behaving correctly. But the problem remains. One more thing to say - when I launch with a sample period equal to 64 (or anything is not equal to 8 and a multiplier of 8) a frame in Webots GUI is a black screen.

ar0usel, 2021-05-03:
I think I advanced in understanding of this problem. If I send the time step is equal to 64 but on every controller step (every 8 ms), It doesn't work. If I send the time step is equal to 64 only when I need information from the camera (every 8 tick of simulation or every 64 ms), It works. 
I guess that the problem is in the part of code, where we enable devices. We re-enable the device with time step more the time step of simulation but every simulation step we tell to device, that we want next measurement after 64 ms. 

P.S. If my explanation is unclear, I will be happy to try to explain more clear.

medrimonia, 2021-05-03:
Hi, there has been a major update in the release branch of the code: https://github.com/RoboCup-Humanoid-TC/webots/
See this commit: https://github.com/RoboCup-Humanoid-TC/webots/commit/aabc4845de623d655b6e86d861d3cd649a9d93fc

Could you try your client with this version and see if it fixes it or not? If it doesn't I'll try to have a more in depth look as soon as possible.

lukicdarkoo, 2021-05-04:
> Blank image (`strlen((char*)rgba_image) = 0`) Moreover, if in the first message the camera time step is equal to 8, it shows one frame and then when I reset time step to 64 in the next message there is no new images.

Note that the `strlen` function returns a number of characters before the first occurrence of 0. Therefore, the function returns 0 if there is no blue in the first pixel.

The best way to verify the image is probably to display it.

ar0usel, 2021-05-04:
> Hi, there has been a major update in the release branch of the code: https://github.com/RoboCup-Humanoid-TC/webots/
> See this commit: [RoboCup-Humanoid-TC@aabc484](https://github.com/RoboCup-Humanoid-TC/webots/commit/aabc4845de623d655b6e86d861d3cd649a9d93fc)
> 
> Could you try your client with this version and see if it fixes it or not? If it doesn't I'll try to have a more in depth look as soon as possible.

Thank you! I checked player.cpp code and looks like the problem with time step not equal to simulation step isn't fixed. 

https://github.com/cyberbotics/webots/blob/a205b1d52306715929119e92f2974909214cf49e/projects/samples/contests/robocup/controllers/player/player.cpp#L491-L520

Here we enable device every simulation step. As I understand we can't do it this way. I guess that this works this way:
1. We try to read the image from camera. (simulation time = 0 ms)
2. We enable camera with time step 64. Simulation return us new image when simulation time will be 64 ms
3. We again try to read image from camera (simulation time = 8 ms)
4. We again enable camera with time step 64. Simulation return us new image when simulation time will be 72 ms
5. And so on 

This mean that we will never get image from camera. I will be grateful, if someone will test this behaviour.

medrimonia, 2021-05-04:
Thanks for the detailed feedback, I will look at this more in depth soon.

medrimonia, 2021-05-04:
It should be fixed in the `experimental` branch of the robocup-hl repository, can you confirm that the fix works?

ar0usel, 2021-05-04:
I checked `experimental` branch, and it doesn't work. Sometimes, in messages that should contain an image, the field `cameras` is empty.

medrimonia, 2021-05-05:
Are you sure you compiled the `player` again? I just tested it using a simple client which reads the message every 64 ms and I was not able to reproduce your bug, it has an image every 64ms. Could you provide more elements for reproducibility?

ar0usel, 2021-05-06:
> It should be fixed in the `experimental` branch of the robocup-hl repository, can you confirm that the fix works?

I tested it one more time and it works. Thank you very much!

closed 2021-05-06 by ar0usel