cyberbotics/webots #3024: youBot mecanumWheel

Thomas-Feldmeier, 2021-04-30:
https://www.kuka.com/en-de/products/mobility/kuka-omnimove-drive-technology:
The wheel consists of two rims and nine free-running rollers, which are mounted at a 45 degree angle

the contactProperties in sampleWorld of youbot are strange, the mecanum wheel angle there is not 45 degrees but something else, so the speed front/back is not same as left/right. i changed them simular to robotino, and it worked. 

change contactProperties in projects/robots/kuka/youbot/worlds/youbot.wbt to this:

contactProperties [
    ContactProperties {
      material1 "InteriorWheelMat"
      coulombFriction [
        0, 2, 0
      ]
      frictionRotation -0.785398 0
      bounce 0
    }
    ContactProperties {
      material1 "ExteriorWheelMat"
      coulombFriction [
        0, 2, 0
      ]
      frictionRotation 0.785398 0
      bounce 0
    }
  ]

the controller could be improved to use something like this (simplified pseudocode of running function)

vx = speed forward/backward, m/s
vy = speed left/right, m/s
vrotation = turning speed, radians/s

move(double vx, vy, vrotation){
  for(int i=0; i<4; i++) {
    motor[i]->setVelocity(
		  (vx
		   + vy * ( (i==1 || i==2) ? 1 : -1)
		   + vrotation * ( (i==1 || i==3) ? 1 : -1) * distanceToRobotCentre)
		  / radius);
  }
}

with motor[0] = left front, motor[1] = right front, motor[2] = left back, motor[3] = right back (watch out, different order than the wheels in youBot)

radius = 0.05m = radius of boundingObject wheels
distanceToRobotCentre = 0.228 + 0.158 = x-offset + y-offset of wheel relative to robot center point

DEF WHEEL1 InteriorWheel {
      translation 0.228 -0.158 -0.055
      anchor 0.228 -0.158 -0.055
      name "wheel1"
      sensorName "wheel1sensor"
    }

Thomas-Feldmeier, 2021-04-30:
in https://cyberbotics.com/doc/guide/youbot should be some documentation about the contactProperties (like in https://cyberbotics.com/doc/guide/robotino3)

mael25, 2021-05-04:
Thank you @Thomas-Feldmeier for raising this issue. The angle of the mecanum wheels has been corrected and the controller improved according to your suggestion in #3025.

closed 2021-05-06 by mael25