cyberbotics/webots #3028: Python controller with multiprocessing Processes will hang the main thread

mezdahun, 2021-05-02:
**Describe the Bug**
I am using Webots 2021a on Windows 10. I am using Thymio2 models with a custom made python controller. Here we would like to continously read sensor values and push these into multiprocessing Queues that will be consumed by parallely running subprocesses. The desired architecture (with multiple processes) is given, as later we would like to set the consumer processes to already implemented ones, but for now we just try to make this basic interface working between webots and multiprocessing processes. 

**Bug:** Even the simplest multiprocessing example will fail to run (hang) when used in the controller code of the robot.

**Example Controller Code:**
```python3
from multiprocessing import Process, Queue, Pool

from controller import Robot

import os

# create the Robot instance.
robot = Robot()

# get the time step of the current world.
timestep = int(robot.getBasicTimeStep())

def setup_sensors():
    global robot
    
    # Creating sensor structure
    sensors = {}
    
    # Setting up proximity sensors
    sensors['prox'] = {}
    sensors['prox']['horizontal'] = []
    
    PROX_UPDATE_FREQ = 10
    for i in range(7):
        device = robot.getDevice(f'prox.horizontal.{i}')
        device.enable(PROX_UPDATE_FREQ)
        sensors['prox']['horizontal'].append(device)        
    return sensors

def test_reader(sensor_stream, motor_stream):
    while True:
        pass

sensors = setup_sensors()

# sensor and motor value queues shared across subprocesses
sensor_stream = Queue()
motor_stream = Queue()

# A process to read and act according to sensor values
reader = Process(target=test_reader, args=(sensor_stream, motor_stream))
reader.start()

# The main thread to interact with non-pickleable objects that can not be passed
# to subprocesses
while robot.step(timestep) != -1:
    prox_vals = [i.getValue() for i in sensors['prox']['horizontal']]
    # theoretically here we pass pickleable sensor values, etc to subprocesses
    # now we just print to see if it is running
    print(prox_vals)
```

**Details:**
as you can see, a Process (in which a loop is running) should be started and this should not block the main thread. In the meanwhile in the main thread we would like to read sensor values. Please note that this is a minimal code without the desired functinality (get and put between queues, etc) just to show that multiprocessing will hang in the controller.

**Behavior:**
After starting the simulation the controller hangs shortly after starting it. Commenting out the process's start method call will resolve the issue. Using `Pool`s and `map_async` or `apply_async` causes the same behavior.

**Steps to Reproduce**
1. Use the attached python code as the controller of a Thymio2 robot in Webots on Windows10 with python3.7

**Expected behavior**
The main thread continously prints the sensor values and an additional process is started.

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce GTX 1650
 - python3.7.9
 - Webots 2021a

**Further notes and questions:**
According to a very [relevant question](https://github.com/cyberbotics/webots/blob/63036ce20cba227d9dd6cbaf8c6f548a8d7e890e/docs/discord/documentation.md#vinwan-11192020-072839) multiprocessing should work out of the box and one should take care of synchronization. Although it is not clear how according to the linked rescources. How can I *"synch everything with `step()`"*?

Could you please let me know if it is possible to use python multiprocessing Processes in a webots controller, and if so, how?

mezdahun, 2021-05-02:
Interestingly, using mutithreading works as expected, so using the following code (where multiprocessing is replaced with threading) will have the expected behavior.

```python3
import threading
from queue import Queue

from controller import Robot

import os

# create the Robot instance.
robot = Robot()

# get the time step of the current world.
timestep = int(robot.getBasicTimeStep())

def setup_sensors():
    global robot
    
    # Creating sensor structure
    sensors = {}
    
    # Setting up proximity sensors
    sensors['prox'] = {}
    sensors['prox']['horizontal'] = []
    
    PROX_UPDATE_FREQ = 10
    for i in range(7):
        device = robot.getDevice(f'prox.horizontal.{i}')
        device.enable(PROX_UPDATE_FREQ)
        sensors['prox']['horizontal'].append(device)
        print(sensors['prox']['horizontal'])
        
    return sensors

def test_reader(sensor_stream, motor_stream):
    while True:
        print('test_reader')
        

sensors = setup_sensors()

# sensor and motor value queues shared across subprocesses
sensor_stream = Queue()
motor_stream = Queue()

# A process to read and act according to sensor values
#reader = Process(target=test_reader, args=(sensor_stream, motor_stream))
#reader.start()

t = threading.Thread(target=test_reader, args=(sensor_stream, motor_stream))
t.start()

# The main thread to interact with non-pickleable objects that can not be passed
# to subprocesses
while robot.step(timestep) != -1:
    prox_vals = [i.getValue() for i in sensors['prox']['horizontal']]
    # theoretically here we pass pickleable sensor values, etc to subprocesses
    # now we just print to see if it is running
    print(prox_vals)
``` 

Could you please let us know, if the same would be possible with multiprocessing?

lukicdarkoo, 2021-05-02:
Can you try keeping your code in functions:
```python

def main():
  # your code

if __name__ == "__main__":
    main()
```

I assume the multiprocessing library tries to serialize the context and it fails as the Webots API is implemented in C/C++ (not serializable).

mezdahun, 2021-05-03:
> Can you try putting your code in the main function (avoid the global space):
> 
> ```python
> def main():
>   # your code
> 
> if __name__ == "__main__":
>     main()
> ```
> 
> I assume the multiprocessing library tries to serialize the context and it fails as the Webots API is implemented in C/C++ (not serializable).

Hi @lukicdarkoo, thank you very much for your quick and helpful reply. Unfortunately using the boilerplate did not solve the issue with using multiprocessing package.

**Changed Snippet:**
```python3
from controller import Robot
from multiprocessing import Queue, Process

# create the Robot instance.
robot = Robot()

# get the time step of the current world.
timestep = int(robot.getBasicTimeStep())

def setup_sensors():
    global robot
    
    # Creating sensor structure
    sensors = {}
    
    # Setting up proximity sensors
    sensors['prox'] = {}
    sensors['prox']['horizontal'] = []
    
    PROX_UPDATE_FREQ = 10
    for i in range(7):
        device = robot.getDevice(f'prox.horizontal.{i}')
        device.enable(PROX_UPDATE_FREQ)
        sensors['prox']['horizontal'].append(device)        
    return sensors

def test_reader(sensor_stream, motor_stream):
    while True:
        print('in test_reader')

def main():
    sensors = setup_sensors()

    # sensor and motor value queues shared across subprocesses
    sensor_stream = Queue()
    motor_stream = Queue()
    
    # A process to read and act according to sensor values
    reader = Process(target=test_reader, args=(sensor_stream, motor_stream))
    reader.start()
    
    # The main thread to interact with non-pickleable objects that can not be passed
    # to subprocesses directly
    while robot.step(timestep) != -1:
        prox_vals = [i.getValue() for i in sensors['prox']['horizontal']]
        # theoretically here we pass pickleable extracted sensor values, etc to subprocesses
        # via Queues now we just print to see if it is running
        print(prox_vals)

if __name__ == "__main__":
    main()
```

**New Behavior:** Unfortunately exatcly the same as before. Starting the Process will hang the controller with no output at all. Commenting out the line when we call `start()` on the process will let the controller get into the main loop continously reading and printing the sensor values. Replacing `multiprocessing.Process` to `threading.Thread` will lead to the desired behavior, i.e. intermittent print of the sensor values from the main thread and some other string from the parallely started process.

Personally this is not a huge problem for our group as we don't really care (during simulations at least) if we use processes or threads, although they have some differences for sure. On the other hand it would be nice to know if it was possible to use multiprocessing as it looks like we are not the only ones trying to do so with python controllers in Webots:

- https://github.com/cyberbotics/webots/issues/2101
- https://github.com/cyberbotics/webots/issues/2057
- https://stackoverflow.com/questions/57132401/how-to-run-a-robots-controller-in-multi-processes-or-multi-threads-in-webots

 If it is not possible atm, It would be great to have maybe just a line in the documentation about this and a possible quickfix with threading if that makes sense? Thanks for the help again, we enjoy webots, keep up the good work!

lukicdarkoo, 2021-05-03:
This works fine for me:

```python
import time
from controller import Robot
from multiprocessing import Queue, Process

def setup_sensors(robot):
    return robot.getDevice('LDS-01')

def test_reader(sensor_stream, motor_stream):
    while True:
        time.sleep(0.1)
        print('in test_reader')

def main():
    robot = Robot()
    timestep = int(robot.getBasicTimeStep())
    sensors = setup_sensors(robot)

    sensor_stream = Queue()
    motor_stream = Queue()
    
    reader = Process(target=test_reader, args=(sensor_stream, motor_stream))
    reader.start()
    
    sensors.enable(timestep)
    while robot.step(timestep) != -1:
        sensors.getRangeImage()
    reader.destroy()

if __name__ == "__main__":
    main()
```

mezdahun, 2021-05-05:
Thank you for this nice working example, it does indeed work for me too after some changes for Thymio2 robots. I was still puzzled though what was the difference that makes the trick. For me the `time.sleep(0.1)` was the one that solved the issue in the subprocess when I used a single `print` statement there.

For anyone who has a similar issue and would have a similar architecture: I created this small snippet according to this help with a single sensor stream and a motor stream. The main thread is pushing values to the sensor stream that can now be passed to python multiprocessing processes and reads from the motor stream where subprocesses can push control motor values. Please note that one should synchronize the subprocesses with the simulation time that is not demonstrated in this snippet. Also, this will only work with Thymio2 robot due to the device names. Also the `time.sleep` is commented out as `multiprocessing.Queue.get` will wait instead.

With this I close the issue, as multiprocessing is indeed compatible with webots. Thanks again @lukicdarkoo !

```python3
import time
from controller import Robot
from multiprocessing import Queue, Process
import logging

# setup logging
logging.basicConfig()
logger = logging.getLogger(__name__)
logger.setLevel('INFO')

def setup_sensors(robot):
    """initialize all available sensors (more info in Thymio2_demo.c controller)
    for this example only a single sensor is returned"""
    return robot.getDevice('prox.horizontal.1')

def setup_motors(robot):
    """initialize available motors (more info in Thymio2_demo.c controller)"""
    motors = {}
    
    motor_left = robot.getDevice("motor.left")
    motor_right = robot.getDevice("motor.right")
    motor_left.setPosition(float('+inf'))
    motor_right.setPosition(float('+inf'))
    motor_left.setVelocity(0)
    motor_right.setVelocity(0)
    
    motors['left'] = motor_left
    motors['right'] = motor_right
    
    return motors

def test_reader(sensor_stream, motor_stream):
    """Method to read sensor values passed from the main thread and act according
        to these by putting motor values to the output stream. This is consumed
        by the main thread and motor values are set accordingly"""
    while True:
        # time.sleep(0.1)
        sensor_value = sensor_stream.get()
        motor_values = {'left': -sensor_value/500, 'right': -sensor_value/500}
        motor_stream.put(motor_values)
        logger.info(f'Pushed motor values {motor_values} according to sensor values in subprocess.')

def main():
    # initialize available webots robot (Thymio2)
    robot = Robot()
    timestep = int(robot.getBasicTimeStep())
    
    # setup and enable devices
    sensors = setup_sensors(robot)
    sensors.enable(timestep)    
    motors = setup_motors(robot)

    # initialize queues to pass and read webots related sensor and motor values
    # in subprocesses
    sensor_stream = Queue()
    motor_stream = Queue()
    
    # create subprocesses to act according to sensor streams
    reader = Process(target=test_reader, args=(sensor_stream, motor_stream))
    reader.start()
    
    # start main thread to read sensor values and act on motor values
    while robot.step(timestep) != -1:
    
        sensor_value = sensors.getValue()
        logger.info(f'Getting sensor value in main thread: {sensor_value}')
        sensor_stream.put(sensor_value)
        
        if motor_stream.qsize() > 0:
            motor_values = motor_stream.get()
            logger.info(f'Setting motor value in main thread: {motor_values}')
            motors['left'].setVelocity(motor_values['left'])
            motors['right'].setVelocity(motor_values['right'])
        
    # cleanup after break
    reader.destroy()
    sensor_stream.close()
    motor_stream.close()

if __name__ == "__main__":
    main()
```

closed 2021-05-05 by mezdahun