cyberbotics/webots #3034: Supervisor API caches positions of nodes

timonegk, 2021-05-05:
**Describe the Bug**
I am using the supervisor API and calling `getPosition()` on nodes to get their position in the world. Now, when the robot is moved in the simulation, the result of `getPosition()` is still the old result even though it has obviously changed.

**Steps to Reproduce**
1. Open Webots
2. Start an external supervisor controller
3. In the controller, get the robot with `robot = supervisor.getFromDef('robot_def')` and the link with `link = robot.getFromProtoDef('link_def')`
4. Call `link.getPosition()`
4. Move the robot with the node in the simulation
5. Step a few times and call `link.getPosition()` again

**Expected behavior**
The returned position should change.

**System**
 - Operating System: Debian Bullseye
 - Graphics Card: –
 - Webots version: 2021b Nightly Build (30-4-2021)

**Additional context**
Add any other context about the problem here.

stefaniapedrazzi, 2021-05-05:
The `getPosition()` function doesn't cache the values but computes them every time from the Webots node.

1) Could you check that it works correctly if you retrieve the position of the 'robot_def' node?
2) Could you make sure that the position of the internal PROTO node 'link_def' is really meant to change?
Note that if you move manually the robot from the scene tree or 3D scene, only the top node position changes and not the internal nodes' position.

timonegk, 2021-05-05:
1. Yes, it does work correctly when `getPosition()` is called on the 'robot_def' node.
2. The robot model is located [here](https://github.com/bit-bots/wolfgang_robot/blob/87df6f7929db59ebdb69590cb7525e033b94a778/wolfgang_webots_sim/protos/Wolfgang.proto).

timonegk, 2021-05-05:
Also, when I call `getVelocity()` on the link node, webots and the controller die without an error message.

stefaniapedrazzi, 2021-05-05:
Yes, the problem is that you are linking to intermediate nested nodes that are just placeholder and are not fully functional.
In fact it works fine for example for `imu_frame` or `r_shoulder` but not for more nested `r_upper_arm`, `r_lower_arm`, etc.

I also suspect that if instead of accessing them directly you pass through the `HingeJointWithBacklash` node first, then it will probably work correctly.

I will check more in detail the issue and inspect how to fix it.

jgueldenstein, 2021-05-05:
I tested this with our robot model before changing the `HingeJoints` to `HingeJointWithBacklash` and the `getPosition()` works correctly.

timonegk, 2021-05-05:
> I also suspect that if instead of accessing them directly you pass through the `HingeJointWithBacklash` node first, then it will probably work correctly.

How would I do that?

stefaniapedrazzi, 2021-05-05:
Instead of getting directly the `r_upper_arm` node from DEF, you should add a DEF name to the `HingeJointWithBacklash`, get its reference with the `getFromProtoDef` function.
Then, from the `HingeJointWithBacklash` node reference call ``getFromProtoDef`` to get the internal endPoint Solid, etc.

You may need to add additional PROTO names or retrieve the nodes using the Supervisor API to get the SFNode/MFNode field, then the child node (by index in case of a MFNode), etc.

PR #3036 (2021-05-10): Recursively search for finalized PROTO node
Fix #3034:
fix the search for PROTO parameter node instance considering only one nested layer.
So accessing parameters of multiple nested PROTO nodes were not correctly handled causing assertion failures and crashes due to request of physical properties to non-finalized intermediate nodes.

This will be the simpler solution for users because they do not need to know the internal sub PROTO node definitions.
However, it is faster to already get the parameter node instance (or the closest node as possible) from the robot controller.