cyberbotics/webots #3063: Translation/Rotation/Velocity not updated inside proto slot inside proto slot

Thomas-Feldmeier, 2021-05-14:
**Describe the Bug**

A lot of robots have extension slots. ToolCenterPoint inside of an vacuum gripper inside of an robot arm does not update position/orientation.

**Steps to Reproduce**

save as Bug.proto in Proto folder:

#VRML_SIM R2021a utf8
PROTO Bug [
  field SFVec3f  translation 0 0 1
  field MFNode children [ ]
]
{
  Transform {
    translation IS translation
    children [
      Transform {
          children IS children
      }
    ]
  }
}

add to a world:
Bug {
  children [
    Bug {
      children [
        Transform {
        }
      ]
    }
  ]
}

change in one of the Bug > translation 1 2 3
click on innermost Transform node, click on Position (absolute), it still shows 0 0 2, not 1 2 4

**System**
Ubuntu 20.04
Webots nightly build 23/04/2021

ad-daniel, 2021-05-14:
I believe this should now be fixed after the very recent changes made in #3036.
Could you try to update to the latest nightly build and verify if indeed the problem persists?

ad-daniel, 2021-06-04:
I've tested the provided example in the latest develop branch and I can't reproduce it. So it's likely to be fixed already and should also be fixed on a recent nightly. I'm closing the issue for now. Feel free to re-open it if something else comes up. Thank you for the report.

closed 2021-06-04 by ad-daniel