cyberbotics/webots #3065: Hang when opening world

ShuffleWire, 2021-05-17:
**Describe the Bug**
When opening a personal world, Webots hang at the step "Finalizing nodes".
Additionally, it's randomly oscillate between 84% and 99%.
Clicking on the mini box of loading seems to trigger the oscillation as well.
More painful, it basically freeze my screen. Drop back to terminal, send SIGKILL to webots, and the process seems to effectively disappear (nothing in htop).
But coming back in GUI, Webots is still here, still hanging and oscillating as before...
Note that the loaded world was working fine before this weekend.
Note that opening another world (which load fine), and then "Open a recent world" allows me to load it successfully.
My world is not that complex, a 4 wheeler style robot, and some OSM data to show surrounding building as box.

**Steps to Reproduce**
1. Open a world (I wish to not share it for the moment, I'm using it at work...)

**Expected behavior**
Loaded the world successfully.

**System**
 - Operating System: Linux Ubuntu 20.04.2 LTS
 - Graphics Card: Not used

lukicdarkoo, 2021-05-17:
Can you identify which commit has caused the error?

ShuffleWire, 2021-05-17:
I don't think it come from a recent commit, and for the moment I mostly blame my computer which was lagging a lot this morning. Rebooted a few times, a now it's ok ! (also performance in simulation increased a lot). Additionally I got the same crash on another model, which was opening fine at the beginning of the day.
Now, I could open and close worlds without issues.
So I do think their is more that a bug in it, but maybe it could be necessary to take a look at the loading code to find out strange behavior ? (but it could be also linked to the handling of the QT windows, which could explain why they keep existing even if I've killed them, and a bug could be more deeply hidden in the interaction with QT).

I'm fine with closing it as it, just want to warn, because personally it would certainly be out of my skills, and I've not much infos to provide...

omichel, 2021-05-18:
OK, let's close the issue for now. Feel free to re-open it if the devil strikes back, hopefully in a reproducible way.

ShuffleWire, 2021-05-18:
Well I've encountered the issue several time now... 
what I've recently made : install a fresh ubuntu 20.04.2.0, update, install git, clone webots and dependancy, make release, then run webots from command line with a empty world in a project directory as webots generate them when using "create new directory" tool.
Hang at 99% when opening. PC not freezed, i'd a twitch stream running in background, and I could still hear the sound. (can't tab out from webots, however)
restart, open the same way, and it work... (2 times)
I'm rather lost

ShuffleWire, 2021-05-19:
I've been able to reproduce the bug each time. I believe that the different popup that appear are random, and only caused by the fact that they show up only after some time. I'm ok with that.

Turn out that the point where everything blows up is  : 
WbApplication.cpp:348:emit postWorldLoaded(reloading, isFirstLoad);
After that, Webots run ok, but if I let gdb run this line, everything explode.
More specificaly, this : 
WbApplication.cpp.moc:385:QMetaObject::activate(this, &staticMetaObject, 1, _a);
Cause the trouble.

Setting a breakpoint on this line (WbApplication.cpp.moc:385) give me :
in : `void WbApplication::postWorldLoaded(bool _t1, bool _t2)`
_t1 = false
_t2 = true
this = (WbApplication * const) 0x5555563558a0
_a = {0x0, 0x7fffffffe1f4, 0x7fffffffe1f0}   // I got 2 times the sames values, is it just luck or not ?
Call stack : 
WbApplication.cpp:348 in `bool WbApplication::loadWorld(QString worldName, bool reloading)` : `emit postWorldLoaded(reloading, isFirstLoad);`
WbMainWindows.cpp:1332 in `bool WbMainWindow::loadWorld(const QString &fileName, bool reloading)` : `const bool success = WbApplication::instance()->loadWorld(fileName, reloading);`
WbGuiApplication.cpp:529 in `void WbGuiApplication::loadInitialWorld()` : `if (!mMainWindow->loadWorld(mStartWorldName))`
WbGuiApplication.cpp:322 in `int WbGuiApplication::exec()` :  `loadInitialWorld();`
main.cpp:207 in `int main(int argc, char *argv[])`  :  `return app.exec();`

If I run one line more, I got the crash.

omichel, 2021-05-19:
Does it crash also if you build Webots with `make release`?

omichel, 2021-05-19:
When you say "Graphics Card: Not used", what does it mean? Your computer doesn't have a graphics card? Or are you using mesa library for OpenGL rendering? Can you give the output of `glxinfo` if available? It could be that the initialization of the OpenGL is too slow and is not complete when Webots attempts to make its first rendering. Also, if you type `webots --sysinfo` what do you get?

ShuffleWire, 2021-05-19:
if `release` is the default target, then yes, it fail also
I've one, but on my Linux install I've never successfully used it...
```./webots --sysinfo
System: Linux 5.8.0-53-generic x86_64
Processor: Intel(R) Core(TM) i9-8950HK CPU @ 2.90GHz
Number of cores: 12
OpenGL vendor: Intel
OpenGL renderer: Mesa Intel(R) UHD Graphics 630 (CFL GT2)
OpenGL version: 4.6 (Compatibility Profile) Mesa 20.2.6```
`glxinfo` is very long... which part do you want ? everything ? Could I use tag to hide it to avoid a very long comment ?

omichel, 2021-05-19:
Intel DRI drivers used to be a little bit buggy, so maybe you have bumped into one of the bugs. If you have a NVIDIA or AMD graphics card onboard, I believe it is worth trying to get it properly installed (software driver for Linux). It is likely to fix the problem.

ShuffleWire, 2021-05-19:
mmm... I've experienced issues when playing around with driver, month ago. Moreover, I don't understand why it's was working fine at some point, and then become more and more crashy... I'll try to find out what happens, but not very sure...

ShuffleWire, 2021-05-20:
So, updating graphic driver made my linux unable to boot up right... Add to remove then and fallback to the opensource one.
I've discovered that the "emit" statement trigger activation of various function (cause of the various "connect", statement).

But actually, running the program under gdb or not make webots hang at two different places, but in the same way as described above (freezing etc...)

With gdb, webots crash here : 
`updateAfterWorldLoading` : 1370 : `mSimulationView->setWorld(WbSimulationWorld::instance());`
`void WbSimulationView::setWorld(WbSimulationWorld *w)`: 802 : `mView3D->setWorld(w);`
`void WbView3D::setWorld(WbSimulationWorld *w)`: 1016 : `enableOptionalRenderingFromPerspective();`
`void WbView3D::enableOptionalRenderingFromPerspective()` : 1140 : `->setChecked(perspective->isGlobalOptionalRenderingEnabled("CoordinateSystem"));`

it's the call to `setChecked` which make webot explode... It very reproducible on my PC. However rarely it **don't** crash if I wait long enough on a breakpoint before running `setChecked`, very strange...

Without gdb, all of this run nicely, and the call to : 
WbGuiApplication.cpp : `int WbGuiApplication::exec()` : 337 : `int ret = QApplication::exec(); ` cause webot to explode the same way.
I think `exec` run various function spread all around the code, to processes event and so on, and the bug could be somewhere in those, but I've no way to track that down (other that putting logs at the beginning and the end of every functions to detect the last one failing, but i don't want to do that...)

A way to get out from this state is to hit Ctrl+Alt+F3 which drop to terminal, login and restart desktop manager (`sudo service gdm3 restart` on my PC), which clean everything without rebooting, but kill every other apps opened...

My solution to all of it : run webots on Debian (11)... Which work very nicely :) ! I do think I will stay with that for the moment...
Maybe Debian totally ignore any graphic card ? I really don't know, at this point.
I'm leaving those infos for those are interested, but I not sure to be able to do much more....

(as a side note, on Debian, I still have to install libjpeg manually, even if this said it was fixed #https://github.com/cyberbotics/webots/issues/1520, is it because I'm building it from source ?)

ShuffleWire, 2021-06-12:
I'm closing as for me, running with Debian fixed the issue. On my PC on Ubuntu, I've got already weird issues regarding display, and it seems to be something related with it. Additionally I can't used my GPU, which could cause another strange behavior.
If anyone get the same sort of issue, feel free to reopen, or link this issue.

closed 2021-06-12 by ShuffleWire