cyberbotics/webots #3072: How does it calculate lift when I use fluid mechanics? In the user's manual, there is only calculate methods of drag force and drag torque.

yuan5, 2021-05-17:
**Documentation page(s) concerned**
Webots reference manual->Nodes and API Functions -> ImmersionProperties

**What is the problem with the current documentation?**
There are equations of drag force and drag torque, however, we want to know how the webots calculate the lift when I use immersion properties simulate(aerodynamics)。

**Describe how you would fix it**
I want know how to calculate lift of wing, or the method to change the force. Thanks!

Best wishes to you!
Crazy Z
2021 05 17

ShuffleWire, 2021-05-25:
Quoting the same doc https://cyberbotics.com/doc/reference/immersionproperties

If the referenceArea is set to "xyz-projected area", the x-coordinate of the drag force vector with respect to the solid frame is given by:

**drag_force_x = - c_x * fluid_density * rel_linear_velocity_x² * sign(rel_linear_velocity_x) * A_x**

where c_x is the x-coordinate of the dragForceCoefficients vector, linear_velocity_x the x-coordinate of the linear velocity 
of the solid with respect to the fluid expressed within the solid frame and A_x is the projected immersed area onto the plane x = 0. 

**Analogous formulas hold for y and z coordinates.** 

Hence, for instance with a plane with the top/down axis being (let's say) the Y axis, then the lift force would be calculated using the same formula with the `y` coordinate.

ShuffleWire, 2021-06-12:
As far as my comprehension goes, the lift is actually the Y coordinate of the so called "drag" vector.  (if the vertical axis of your plane is the Y axis). The actual drag force would be the X coordinate of the "drag vector" (if the front-back axis of your place is the X axis).
I don't think their is more bhind that, the lift and drag force are actually computed in very similar way.

I do agree that "drag" is a confusing term. If my explanation is right, then maybe should we discuss about changing is name ? Or at least mention it in the documentation (in order to avoid complex renaming issue)

ShuffleWire, 2021-07-27:
After having think about it, my thought is not right, because the force in the Y axis is caused by the speed along this axis, not the speed along the X axis, as the lift would normal do. Not sure, however...

yuan5, 2021-07-27:
OK, thank you for your answer. I found a new way to do hydrodynamics simulation in webots. We get the relationship between aerodynamics and motion of solid. Then, we add the force by supervisor. The function wb_supervisor_node_add_force() & wb_supervisor_node_add_force_with_offset().

closed 2021-07-27 by yuan5