cyberbotics/webots #3078: Crash when deleting robot

omichel, 2021-05-18:
It seems that if a supervisor process is monitoring contact points of robots with getContactPoints() and deletes a robot with remove(), webots crashes with the following stack:
```
Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007ff7c62ec90c in WbOdeGeomData::solid (this=0xfeeefeeefeeefeee)
    at ode/WbOdeGeomData.hpp:39
39        WbSolid *solid() const { return mSolid; }
(gdb) where
#0  0x00007ff7c62ec90c in WbOdeGeomData::solid (this=0xfeeefeeefeeefeee)
    at ode/WbOdeGeomData.hpp:39
#1  0x00007ff7c607d366 in WbSolid::extractContactPoints (this=0x1e0637d8bd0)
    at nodes/WbSolid.cpp:2411
#2  0x00007ff7c607d17c in WbSolid::computedContactPoints (this=0x1e0637d8bd0,
    includeDescendants=false) at nodes/WbSolid.cpp:2385
#3  0x00007ff7c609a469 in WbSupervisorUtilities::writeAnswer (this=
    0x1e051a40e60, stream=...) at nodes/utils/WbSupervisorUtilities.cpp:1563
#4  0x00007ff7c6041857 in WbRobot::writeAnswer (this=0x1e05160db20,
    stream=...) at nodes/WbRobot.cpp:1221
#5  0x00007ff7c60405af in WbRobot::dispatchAnswer (this=0x1e05160db20,
    stream=..., includeDevices=true) at nodes/WbRobot.cpp:1047
#6  0x00007ff7c5f47c0a in WbController::writeAnswer (this=0x1e04d9e0cb0,
    immediateAnswer=true) at control/WbController.cpp:941
#7  0x00007ff7c5f48337 in WbController::readRequest (this=0x1e04d9e0cb0)
    at control/WbController.cpp:1092
#8  0x00007ff7c6184c47 in WbController::qt_static_metacall (_o=0x1e04d9e0cb0,
    _c=QMetaObject::InvokeMetaMethod, _id=2, _a=0x58fb1fbc20)
    at build/debug/WbController.moc.cpp:147
#9  0x00007ffb52eaf203 in void doActivate<false>(QObject*, int, void**) ()
   from D:\msys64\mingw64\bin\Qt5Core.dll
#10 0x00007ffb52eaf203 in void doActivate<false>(QObject*, int, void**) ()
   from D:\msys64\mingw64\bin\Qt5Core.dll
#11 0x00007ffb52d74f62 in QWindowsPipeReader::emitPendingReadyRead() ()
   from D:\msys64\mingw64\bin\Qt5Core.dll
#12 0x00007ffb52ded782 in QMetaCallEvent::placeMetaCall(QObject*) ()
   from D:\msys64\mingw64\bin\Qt5Core.dll
#13 0x00007ffb52df272e in QObject::event(QEvent*) ()
   from D:\msys64\mingw64\bin\Qt5Core.dll
#14 0x00007ffb537d7fd3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from D:\msys64\mingw64\bin\Qt5Widgets.dll
#15 0x00007ffb52dc4aca in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from D:\msys64\mingw64\bin\Qt5Core.dll
#16 0x00007ffb52dcb119 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from D:\msys64\mingw64\bin\Qt5Core.dll
#17 0x00007ffb5168195e in QWindowsGuiEventDispatcher::sendPostedEvents() ()
   from D:\msys64\mingw64\share\qt5\plugins\platforms\qwindows.dll
#18 0x00007ffb52e2128f in QEventDispatcherWin32::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from D:\msys64\mingw64\bin\Qt5Core.dll
#19 0x00007ffb51681945 in QWindowsGuiEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from D:\msys64\mingw64\share\qt5\plugins\platforms\qwindows.dll
#20 0x00007ffb52dc3473 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from D:\msys64\mingw64\bin\Qt5Core.dll
#21 0x00007ffb52dcc127 in QCoreApplication::exec() ()
   from D:\msys64\mingw64\bin\Qt5Core.dll
#22 0x00007ff7c5f825f2 in WbGuiApplication::exec (this=0x58fb1ff560)
    at gui/WbGuiApplication.cpp:341
#23 0x00007ff7c6107ff3 in main (argc=1, argv=0x1e03a704240)
    at gui/main.cpp:207
```

ShuffleWire, 2021-05-18:
Do you have a faulty world to share ?
For me `0xfeeefeeefeeefeee` is definitely not a valid address, anyway.

omichel, 2022-06-17:
I cannot reproduce this crash with the current version of Webots on develop.