cyberbotics/webots #3091: Stuck on "Loading World" - R2021a

UnleashedGH, 2021-05-25:
**Describe the Bug**
Webots gets stuck on "Loading World" after running the program.

**Steps to Reproduce**
1- Run Webots
2-Stuck on "Loading World", nothing is rendering

**Expected behavior**
For the scene to render

**Screenshots**
[image]
(I get weird artifacts when I move windows around, seems that the viewport doesn't refresh correctly

**System**
 - Operating System: Windows 10 - 64 Bit
 - Graphics Card: Intel (R) HD Graphics

**Additional context**
It's worth noting that there is no CPU usage when "Loading World" appears, it doesn't seem that the program is actually loading anything but rather crashing, maybe not the entire program but at least the view port seems so.
objects also appear in the Scene Tree.

I followed the troubleshooting guide in regards to running in Safe Mode, with that I got the view port to render; I was able to add a background with a blue sky color , along with a table object; I was able to move, rotate, and zoom the camera with no issues

however, when I get to Wizard -> New Project Directory and create a new project, I get stuck on "Loading World" once again.
and also the objects appear in the Scene Tree,

omichel, 2021-05-25:
This problem is likely due to your GPU which is below the recommended requirements for Webots. Can you try it on a computer with a NVIDIA GPU?

UnleashedGH, 2021-05-25:
> This problem is likely due to your GPU which is below the recommended requirements for Webots. Can you try it on a computer

Only have access to this one.
I also have all OpenGL settings set to absolute lowest
[image]

I understand that its quite an extreme stress test, but I'm just confused on why the scene would render and function correctly when I'm on safe mode and no project opened
no lag or framerate issues either

ShuffleWire, 2021-05-25:
I'm taking the opportunity to link this https://github.com/cyberbotics/webots/issues/3065 issue, don't know if it's related in anyway, but still.

UnleashedGH, 2021-05-26:
> This problem is likely due to your GPU which is below the recommended requirements for Webots. Can you try it on a computer with a NVIDIA GPU?

I was able to narrow down the issue, it's because of the **TexturedBackground (Background)** Node
once I add it in a scene, even if its completely empty, the render crashes (white screen and other graphical artifacts)
[image]

however, if i just add a **TexturedBackgroundLight (DirectionalLight)** instead, scene renders correctly with no issues
i can add objects, move the camera, etc
[image]

the problem is that Wizard -> New Project Directory adds the Node by default, is there a way I can workaround that to prevent it from being added?

ShuffleWire, 2021-05-26:
You can untick it in the world creation window : 
[image]

lukicdarkoo, 2021-05-26:
> * Graphics Card: Intel (R) HD Graphics

What is the generation of the Intel CPU?

ShuffleWire, 2021-05-26:
> > * Graphics Card: Intel (R) HD Graphics
> 
> What is the generation of the Intel CPU?

I wish to say that even on a very old CPU (on my laptop I've a Intel 2020M, with onboard graphic being very old), it work nicely (not saying it's fast by any mean, but stable and functional :) )
Thus, I'm not sure if the integrated graphic are to blame, here. But hardware 3D rendering could be a mess, still...

UnleashedGH, 2021-05-26:
> You can untick it in the world creation window :
> [image]

Wow, I completely missed those options
Thank you

> > * Graphics Card: Intel (R) HD Graphics
> 
> What is the generation of the Intel CPU?

Intel(R) Celeron(R) CPU N3060 @ 1.60Hz

lukicdarkoo, 2021-05-26:
> I wish to say that even on a very old CPU (on my laptop I've a Intel 2020M, with onboard graphic being very old), it work nicely (not saying it's fast by any mean, but stable and functional :) ) [...]

I suspected that some OpenGL features are not present in the old integrated GPUs, but this should not be the case for this given CPU:
https://en.wikichip.org/wiki/intel/celeron/n3060

omichel, 2021-05-26:
In any case, these old integrated GPUs are known to have random OpenGL bugs and this is why they are not officially supported by Webots.

ShuffleWire, 2021-05-26:
> In any case, these old integrated GPUs are known to have random OpenGL bugs and this is why they are not officially supported by Webots.

It make sense, as far as I've experienced, it work, but sometimes it get crazy

ShuffleWire, 2021-05-26:
@UnleashedGH did unticking "textured background" fix your issue ?

UnleashedGH, 2021-05-26:
> @UnleashedGH did unticking "textured background" fix your issue ?

yes

ShuffleWire, 2021-06-12:
As the solution seems to be a sufficient workaround, and the question related to non--GPU usage, we might close this issue, @omichel ?

closed 2021-06-12 by ShuffleWire