cyberbotics/webots #3094: Mismatching tag in connector documentation

ShuffleWire, 2021-05-26:
**Documentation page(s) concerned**
https://www.cyberbotics.com/doc/reference/connector

**What is the problem with the current documentation?**
The link "this figure" in "Field Summary" don't actually link to the figure, because the tag of the figure is wrong
[image]

**Describe how you would fix it**
It would just be necessary to change the tag of the image from `#example-of-rotational-alignment-numberofrotations4-and-rotationaltolerance22-5-deg` to `#example-of-rotational-alignment` , but I don't know how/where to do that, if anyone could drive me :)

ShuffleWire, 2021-05-26:
Should I change 

`[this figure](#example-of-rotational-alignment)`
to 
`[this figure](#example-of-rotational-alignment-numberofrotations4-and-rotationaltolerance22-5-deg)`

or 

`%figure "Example of rotational alignment (numberOfRotations=4 and rotationalTolerance=22.5 deg)"`
to
`%figure "Example of rotational alignment"`

Changing the first force to make a strange looking tag, but changing the second will remove interesting information for the reader.

ShuffleWire, 2021-05-26:
BTW, what the difference between `released` and `master` ?

stefaniapedrazzi, 2021-05-26:
`released` is only used for documentation fixes.

stefaniapedrazzi, 2021-05-26:
Thank you!

I will now check and fix the test error.

stefaniapedrazzi, 2021-05-27:
Fixed in #3095

closed 2021-05-27 by stefaniapedrazzi