cyberbotics/webots #3127: Test suite is broken under Linux

omichel, 2021-06-02:
On several PRs, the Linux test suite is passing, but returns a wrong exit code:
```
Test suite complete
Error: Process completed with exit code 1.
```
See for example https://github.com/cyberbotics/webots/runs/2728593086

ad-daniel, 2021-06-04:
Appears to be related to expired keys

```
  Hit:1 http://azure.archive.ubuntu.com/ubuntu bionic InRelease
  Hit:2 http://azure.archive.ubuntu.com/ubuntu bionic-updates InRelease
  Hit:3 http://azure.archive.ubuntu.com/ubuntu bionic-backports InRelease
  Hit:4 https://packages.microsoft.com/ubuntu/18.04/prod bionic InRelease
  Hit:5 http://security.ubuntu.com/ubuntu bionic-security InRelease
  Hit:6 http://ppa.launchpad.net/ondrej/php/ubuntu bionic InRelease
  Get:7 http://packages.ros.org/ros/ubuntu bionic InRelease [4,680 B]
  Hit:8 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu bionic InRelease
  Get:9 http://packages.ros.org/ros2/ubuntu bionic InRelease [4,680 B]
  Err:7 http://packages.ros.org/ros/ubuntu bionic InRelease
    The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
  Err:9 http://packages.ros.org/ros2/ubuntu bionic InRelease
    The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
  Reading package lists...
  W: GPG error: http://packages.ros.org/ros/ubuntu bionic InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
  E: The repository 'http://packages.ros.org/ros/ubuntu bionic InRelease' is not signed.
  W: GPG error: http://packages.ros.org/ros2/ubuntu bionic InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
  E: The repository 'http://packages.ros.org/ros2/ubuntu bionic InRelease' is not signed.
Error: The process '/usr/bin/sudo' failed with exit code 100
```
and be related to [this](https://discourse.ros.org/t/ros-gpg-key-expiration-incident/20669)

lukicdarkoo, 2021-06-04:
A few days ago they messed up the keys, so it is probably the reason

omichel, 2021-06-04:
Can we expect they will resolve it on their end soon?

omichel, 2021-06-07:
It seems to be resolved by now.

closed 2021-06-07 by omichel