cyberbotics/webots #3132: Crash in MATLAB launcher on develop branch

mael25, 2021-06-03:
**Describe the Bug**
The launcher crash when trying to to get the environment of the project on the `develop` branch.

**Steps to Reproduce**
1. Launch any MATLAB controller

**Expected behavior**
Manage to find the environment path.

**Screenshots**
[image]

**Additional context**
The function `wbu_system_getenv` does not exist anymore in `lib/controller/matlab`.

mael25, 2021-06-03:
It works well when replacing l.156-157 of `lib/controller/matlab/launcher.m` with:
```
  WEBOTS_PROJECT_UTF8 = calllib('libController', 'wbu_system_getenv', 'WEBOTS_PROJECT');
  WEBOTS_PROJECT = calllib('libController', 'wbu_system_short_path', WEBOTS_PROJECT_UTF8);
```

But then other functions are not defined for MATLAB:
[image]

So there must be an issue with the adaptation of the functions for MATLAB.

BenjaminDeleze, 2021-06-04:
Do you have the `wbu_system_getenv.m` file in `webots/lib/controller/matlab`?
Are you on Windows or Linux?

mael25, 2021-06-04:
No, it is note ther anymore. I use Linux.

BenjaminDeleze, 2021-06-04:
Can you try to `make` and see if it is still missing please?

mael25, 2021-06-04:
Yes, it seems to still be missing...

BenjaminDeleze, 2021-06-04:
ok, I cannot reproduce it.
In your make process, did you have these lines printed
` Matlab controller library (release)  
generating matlab
Unable to find python, skipping matlab API`

it should appear between:
` Java controller library (release)`

and 

` Python 2.7 controller library (release) `

mael25, 2021-06-04:
Yes I have `Unable to find python, skipping matlab API` , so I think we can close this issue.

BenjaminDeleze, 2021-06-04:
I opened a PR (#3137) about that.

Can you test it to see if it fixes the problem?

mael25, 2021-06-04:
Solved by #3137

closed 2021-06-04 by mael25