cyberbotics/webots #3133: --minimize option not working on Ubuntu 18.04

stefaniapedrazzi, 2021-06-03:
In the develop branch R2021b the `--minimize` option is no longer working on Ubuntu 18.04.
Even if executing `./wetbos --minimize`, the Webots window is displayed.

lukicdarkoo, 2021-06-03:
I use it quite often and it works on Ubuntu 20.04, even with different desktop environments.

omichel, 2021-06-04:
I just tested on my Ubuntu 18.04, current develop branch and the `--minimize` option works as expected.

omichel, 2021-06-04:
It could be related to this: https://bugreports.qt.io/browse/QTBUG-76354

stefaniapedrazzi, 2021-06-09:
Without recompiling anything but just restarting the PC, I could not reproduce the issue any longer.
But since a couple of weeks I noticed this issue multiple times, especially when running the test suite.
I assigned it to me and I will monitor this issue until the release.

stefaniapedrazzi, 2021-07-01:
I could reproduce it also with a nightly build package.
But just once, the first time, and then it worked correctly.

stefaniapedrazzi, 2021-07-12:
I tested with the R2021b release candidate tarball package and still I could reproduce this issue on Ubuntu 18.04.
It is not happening every time, but sometimes it occurs that the minimize option is not applied.
I still didn't find a procedure to reproduce it systematically.

omichel, 2022-06-13:
Since we are dropping Ubuntu 18.04 support, I believe we can close this issue.

closed 2022-06-13 by omichel