cyberbotics/webots #3138: use GPS function "wb_gps_get_values" in simulink matlab function

qingliuql, 2021-06-05:
**Describe the Bug**

Undefined function or variable 'setdatatype'.
Function 'wb_gps_get_values.m' (#128.284.318), line 8, column 1:
"setdatatype(obj,'doublePtr', 1, 3)"
Launch diagnostic report.
Component: MATLAB Function | Category: Coder error
Function call failed.
Function 'MATLAB Function' (#122.36.58), line 3, column 8:
"wb_gps_get_values(gps)"

**Steps to Reproduce**
1. Go to 'webots2020b-rev2 one world'; The world has one robot which has gps sensor in the centroid position;
2. the robot controller is one m file and one simulink file;
2. Click on 'Run the simulation in real time' and open_system('wheelleg_simulink');
3. When the system wheelleg_simulink is open, run the simulink file;
4. See error:
    Function 'wb_gps_get_values.m' (#128.284.318), line 8, column 1:
   "setdatatype(obj,'doublePtr', 1, 3)"

**Expected behavior**
I hope I can use the GPS function in simulink, because model the controller of robot in simulink is easy and efficient.

**Screenshots**
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce GTX 950, 
 -matlab2021a & webots2020b-rev2

lukicdarkoo, 2021-06-07:
Can you try R2021a? If that doesn't work then R2021b? Webots 2020b-rev2 is an old version and we might have solved those problems in the newer version.

qingliuql, 2021-06-09:
> Can you try R2021a? If that doesn't work then R2021b? Webots 2020b-rev2 is an old version and we might have solved those problems in the newer version.

Ok, thank you! I will try it later.

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel