cyberbotics/webots #3139: getRecognitionObjects() method not recognized for Java

mreliotylhs, 2021-06-06:
**Describe the Bug**
the Java method **getRecognitionObjects()** is not recognized and does not appear to work.  All other methods work except this critical method.

**Steps to Reproduce**
Using Java as the Controller type

(1)  Enable Recognition
(2)  Create A Camera, such as:
    Camera MyCamera = MyRobot.getCamera("camera");
    MyCamera.enable(timeStep);
    MyCamera.recognitionEnable(timeStep);

(3)  Attempt to get Recognition Objects
    if(MyCamera.hasRecognition() == true)
        CameraRecognitionObject[] MyObjects = MyCamera.getRecognitionObjects();

**Expected behavior**
An array of Recognition objects is stored, and the method is recognized.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: MacOS 
 - Graphics Card: Unrelated

**Additional context**
I have tested all other Camera and Recognition class methods, they work.

stefaniapedrazzi, 2021-06-16:
I just checked and it works correctly for me.
Which version of Java are you using?
Could you please try to run this controller?
If it doesn't work, could you please report the error messages you get.

``` java
import com.cyberbotics.webots.controller.Robot;
import com.cyberbotics.webots.controller.Camera;
import com.cyberbotics.webots.controller.CameraRecognitionObject;

public class CameraRecognition {

  public static void main(String[] args) {
    Robot robot = new Robot();

    int timeStep = (int) Math.round(robot.getBasicTimeStep());
    
    Camera camera = robot.getCamera("camera");
    camera.enable(timeStep);
    camera.recognitionEnable(timeStep);
    
    while (robot.step(timeStep) != -1) {
      int n = camera.getRecognitionNumberOfObjects();
      System.out.println("Number of objects: " + n);
      
      CameraRecognitionObject[] objects = camera.getRecognitionObjects();
      for (CameraRecognitionObject obj : objects) {
        System.out.println("  Object: " + obj.getModel());
        System.out.println("    id: " + obj.getId());
        System.out.println("    relative position: " + obj.getPosition()[0] + " " + obj.getPosition()[1] + " " + obj.getPosition()[2]);
        System.out.println("    size: " + obj.getSize()[0] + " " + obj.getSize()[1]);
        System.out.println("    positoin on image: " + obj.getPositionOnImage()[0] + " " + obj.getPositionOnImage()[1]);
        System.out.println("    size on image: " + obj.getSizeOnImage()[0] + " " + obj.getSizeOnImage()[1]);
        for (int i = 0; i < obj.getNumberOfColors(); i++)
          System.out.println("    color " + (i+1) + ": " + obj.getColors()[3 * i] + " " + obj.getColors()[3 * i + 1] + " " + obj.getColors()[3 * i + 2]);
        
      }
    }
  }
}
```

Here you can find a world you could use to test the controller:
https://github.com/cyberbotics/webots/blob/master/projects/samples/devices/worlds/camera_recognition.wbt

stefaniapedrazzi, 2021-06-17:
We recently fixed some related issues, so I would suggest you to try the latest nightly build:
https://github.com/cyberbotics/webots/releases/tag/nightly_16_6_2021

mreliotylhs, 2021-06-17:
Here is the specific error I am getting:

[image: image.png]

Looking at your second email, it looks like the renaming of the Java method
in the nightly build has solved the problem.

In version *2021a of Webots*, the method is named "
getCameraRecognitionObjects()" whereas the Camera API for Java shows it as "
getRecognitionObjects()"

[image: image.png]

So yea, that's the problem and the nightly build solves it.

Dan

mreliotylhs, 2021-06-17:
OK,

Problem solved based on a reading of the changes in the 06/16/2021 nightly build.

Here is the error I am getting in Version 2021a stable:

[image]

In version 2021a of Webots, the method is actually named "**getCameraRecognitionObjects()**" whereas the Camera API for Java shows it as "**getRecognitionObjects()**".  Since the API was wrong, I was getting the error.

[image]

So adding "Camera" works for the current stable version, or I can upgrade to the nightly build and use the "correct" method name from the API of getRecognitionObjects()

closed 2021-06-18 by stefaniapedrazzi