cyberbotics/webots #3158: Python controller segfaults in RangeFinder::__getRangeImageList

sigmunau, 2021-06-10:
**Describe the Bug**
When running a python controller that uses threads (even if the other threads does not touch webots api) the controller process segfaults in PyFloat_FromDouble called from __getRangeImageList
Tried on both https://github.com/cyberbotics/webots/releases/download/nightly_9_6_2021/webots_2021a-rev1_amd64.deb and https://github.com/cyberbotics/webots/releases/download/nightly_9_6_2021/webots_2021b_amd64.deb with seemingly the same results. The stacktrace below is from 2021b:

Most likely this is caused by a previous change to release python GIL while in c++ code. This also (seemingly) releases the log inside the %extends blocks which may call non-threadsafe python functions. The solution to this seems to be to wrap such code like this
```
%nothread;
<extension code>
%thread;
```
gdb stack trace:
```
#0  0x00000000005bc013 in PyObject_Malloc ()
[Current thread is 1 (Thread 0x7fdc41323740 (LWP 79169))]
(gdb) bt
#0  0x00000000005bc013 in PyObject_Malloc ()
#1  0x00000000005ed03e in PyFloat_FromDouble ()
#2  0x00007fdc3fbd84c8 in webots_RangeFinder___getRangeImageList(webots::RangeFinder*) () from /usr/local/webots/lib/controller/python38/_controller.so
#3  0x00007fdc3fc100a8 in _wrap_RangeFinder___getRangeImageList ()
   from /usr/local/webots/lib/controller/python38/_controller.so

```

**Steps to Reproduce**
Load world file
Start controller
Controller crashes when it calls RangeFinder.getRangeImage('list')

**Expected behavior**
Controller does not crash

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
Ubuntu 20.04 webots nightly 20
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

omichel, 2021-06-11:
Does adding:
```
%nothread __getRangeImageList;
```
right after:
```
%module(threads=1) controller
```
fixes the problem for you?

sigmunau, 2021-06-11:
I think there are many functions which could cause problems, this is just the one where it is most likely to happen since it allocates a lot of python objects (which, as a side note made me realize that we really wanted to use `getRangeImage('buffer')` which "fixes" the problem. I tried to add some % nothread in that file , but I tried it on the develop branch (as you asked for last time), but that branch doesn't work for me at all. Looks like the axis' are all mixed up

omichel, 2021-06-11:
After switching from the master to the develop branch, did you rebuilt everything? If the problem persists, what do you mean exactly with the axis are all mixed up?

sigmunau, 2021-06-14:
I ran make clean and make. This should be enough, right? I could send a screenshot showing the problem, but I'd prefer to not do it in a public github issue

omichel, 2021-06-14:
Please send your screenshot to support@cyberbotics.com, we will keep it confidential.

PR #3389 (2021-07-12): Acquire the python GIL for all functions except Robot.step()
Fixes #3158 and replaces #3176.

It seems to be a cleaner path than #3176.
closed 2021-07-12 by omichel