cyberbotics/webots #3165: Displayed speed is not correct

ShuffleWire, 2021-06-11:
**Describe the Bug**
The displayed velocity in the "Velocity tab" isn't correct.

**Steps to Reproduce**
1. Open the following world :
<details>
<summary>world file</summary>
<pre>
#VRML_SIM R2021a utf8
WorldInfo {
  coordinateSystem "NUE"
  lineScale 10
}
Viewpoint {
  orientation -0.9351235864913506 -0.31827338836034974 -0.1557110408711453 0.9640363489590187
  position -1.87889558080866 7.2834176780814746 6.396411931107234
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Robot {
  children [
    HingeJoint {
      jointParameters HingeJointParameters {
        axis 0 1 0
      }
      device [
        RotationalMotor {
        }
      ]
      endPoint Solid {
        children [
          Solid {
            translation 2 0 0
            children [
              Shape {
                geometry Cylinder {
                }
              }
            ]
          }
          Shape {
            geometry Box {
            }
          }
        ]
        name "solid(1)"
      }
    }
  ]
  controller "my_controller"
}
</pre>
</details>

2. Create a controller named my_controller and use the following code :
<details>
<summary>controller file</summary>
<pre>
#include &lt;webots/robot.h&gt;
#include &lt;webots/motor.h&gt;

#define TIME_STEP 64
int main(int argc, char **argv) {
  wb_robot_init();
  WbDeviceTag motor = wb_robot_get_device("rotational motor");
while (wb_robot_step(TIME_STEP) != -1) {
wb_motor_set_position(motor, INFINITY);
wb_motor_set_velocity(motor, M_PI/4);
};
  wb_robot_cleanup();
  return 0;
}
</pre>
</details>

3. Run simulation
4. Go to any Solid node, and see that the speed in the "Velocity" tab (linear and rotation, whatever the depth level) is zero, even if the object actually move.

**Expected behavior**
The displayed speed should be the true speed.

**System**
 - Operating System: Debian 11

ShuffleWire, 2021-06-18:
Just realized that I need a Physic node for each node where I want to read velocity.
It actually fix this issue, but I not comfortable with it : Position is nicely updated, but not velocity, with is confusing. I assume that is because velocity is an output of the ode solver.
Any way, adding a physic node just to get object velocity doesn't feel right, especially because adding many physic node will have a negative impact on performances.

omichel, 2021-06-18:
Unfortunately, the kinematic engine in Webots is not very elaborated and could be easily improved with respect to this kind of things.

ShuffleWire, 2021-06-18:
Does the kinematic engine is internal to webots ? or it's one of the dependency ?
Additionally, is it the same engine which is not able to use close loop kinematic (ie, without physic node) ? If so, maybe I could have a look to try to rework it.

omichel, 2021-06-18:
> Does the kinematic engine is internal to webots ? or it's one of the dependency ?

It's internal.

>  Additionally, is it the same engine which is not able to use close loop kinematic (ie, without physic node) ?

Yes.

ShuffleWire, 2021-06-21:
Ok, then I'm closing the issue for the time being. Maybe I will spend some time later to work on the engine, later.

closed 2021-06-21 by ShuffleWire