cyberbotics/webots #3169: Crash when resetting something already at the default value

ShuffleWire, 2021-06-12:
**Describe the Bug**
When resetting something already at the default value, Webots crash in certains circumstances.

**Steps to Reproduce**
1. Create a new world
2. Run a few second (don't know if runing actually as a influence or not, nor if the duration as any impact)
3. Add a LensFlare in Viewpoint. Actually, It work with a lot of thing (a bounding box / physic in a robot, a PRBApparence in a Shape... I think everything that is actually to a NULL value at the start)
4. Run a few second (don't know if runing actually as a influence or not, nor if the duration as any impact)
5. Pause
6. Reset the world
7. See that the object previously set have disappear (which is actually a not so convenient feature, I must say, but still...)
8. Right click on it, The option "Reset to default value is available", which is a bug in itself. (got it each time)
9. Use this option, sometime Webots crash. During my experiment I had to try sometimes multiple stuff like that before getting the crash, sometime I got it right away.

**Traceback**

End up on a SegFault.

<details>
<summary>Traceback</summary>
<pre>
#0  0x00007ffff790fd19 in operator==(QString const&, QString const&) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#1  0x0000555555a8572f in WbNodeEditor::apply() (this=0x555556baa970) at scene_tree/WbNodeEditor.cpp:175
#2  0x0000555555a8d718 in WbNodePane::apply() (this=0x5555567cfff0) at scene_tree/WbNodePane.cpp:142
#3  0x00005555559dcf71 in WbValueEditor::applyIfNeeded() (this=0x5555567cfff0) at scene_tree/WbValueEditor.hpp:72
#4  0x00005555559dc724 in WbFieldEditor::editField(WbNode*, WbField*, int) (this=0x55555673da90, node=0x0, field=0x0, item=-1) at scene_tree/WbFieldEditor.cpp:199
#5  0x0000555555afcf9d in WbSceneTree::reset() (this=0x555556abc350) at scene_tree/WbSceneTree.cpp:519
#6  0x0000555555b05103 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbSceneTree::*)()>::call(void (WbSceneTree::*)(), WbSceneTree*, void**)
    (f=(void (WbSceneTree::*)(WbSceneTree * const)) 0x555555afccaa <WbSceneTree::reset()>, o=0x555556abc350, arg=0x7fffffffbeb0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:152
#7  0x0000555555b04de9 in QtPrivate::FunctionPointer<void (WbSceneTree::*)()>::call<QtPrivate::List<>, void>(void (WbSceneTree::*)(), WbSceneTree*, void**)
    (f=(void (WbSceneTree::*)(WbSceneTree * const)) 0x555555afccaa <WbSceneTree::reset()>, o=0x555556abc350, arg=0x7fffffffbeb0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:185
#8  0x0000555555b044c0 in QtPrivate::QSlotObject<void (WbSceneTree::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*)
    (which=1, this_=0x555556a52060, r=0x555556abc350, a=0x7fffffffbeb0, ret=0x0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:418
#9  0x00007ffff7a9ff30 in void doActivate<false>(QObject*, int, void**) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#10 0x00007ffff6154672 in QAction::triggered(bool) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#11 0x00007ffff6156a9e in QAction::activate(QAction::ActionEvent) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#12 0x00007ffff62c653a in QMenuPrivate::activateCausedStack(QVector<QPointer<QWidget> > const&, QAction*, QAction::ActionEvent, bool) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#13 0x00007ffff62cd666 in QMenuPrivate::activateAction(QAction*, QAction::ActionEvent, bool) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#14 0x00007ffff62ce7c0 in QMenu::mouseReleaseEvent(QMouseEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#15 0x00007ffff6198740 in QWidget::event(QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#16 0x00007ffff62d0cdb in QMenu::event(QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#17 0x00007ffff615a43c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#18 0x00007ffff61611f8 in QApplication::notify(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#19 0x00007ffff7a67808 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#20 0x00007ffff616053a in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
    at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#21 0x00007ffff61b1cda in QWidgetWindow::handleMouseEvent(QMouseEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#22 0x00007ffff61b4cf3 in QWidgetWindow::event(QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#23 0x00007ffff615a43c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#24 0x00007ffff6160f20 in QApplication::notify(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#25 0x00007ffff7a67808 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#26 0x00007ffff6c5a56d in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#27 0x00007ffff6c5b955 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#28 0x00007ffff6c378ab in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#29 0x00007fffe6c9d69a in xcbSourceDispatch(_GSource*, int (*)(void*), void*) () at /home/yann/webots/lib/webots/libQt5XcbQpa.so.5
#30 0x00007ffff1513e6b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff1514118 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff15141cf in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff7ac31cc in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#34 0x00007ffff7a6621a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#35 0x00007ffff62cb3e4 in QMenuPrivate::exec(QPoint const&, QAction*, std::function<QPoint (QSize const&)>) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#36 0x00007ffff62cb4db in QMenu::exec(QPoint const&, QAction*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#37 0x0000555555c08e1f in WbContextMenuGenerator::generateContextMenu(QPoint const&, WbNode const*) (position=..., selectedNode=0x0) at user_commands/WbContextMenuGenerator.cpp:166
#38 0x0000555555ba1aaa in WbTreeView::showMenu(QPoint const&) (this=0x55555c38d0c0, position=...) at scene_tree/WbTreeView.cpp:128
#39 0x0000555555ba2460 in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QPoint const&>, void, void (WbTreeView::*)(QPoint const&)>::call(void (WbTreeView::*)(QPoint const&), WbTreeView*, void**) (f=(void (WbTreeView::*)(WbTreeView * const, const QPoint &)) 0x555555ba19ec <WbTreeView::showMenu(QPoint const&)>, o=0x55555c38d0c0, arg=0x7fffffffcfb0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:152
#40 0x0000555555ba23e3 in QtPrivate::FunctionPointer<void (WbTreeView::*)(QPoint const&)>::call<QtPrivate::List<QPoint const&>, void>(void (WbTreeView::*)(QPoint const&), WbTreeView*, void**) (f=(void (WbTreeView::*)(WbTreeView * const, const QPoint &)) 0x555555ba19ec <WbTreeView::showMenu(QPoint const&)>, o=0x55555c38d0c0, arg=0x7fffffffcfb0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:185
#41 0x0000555555ba234e in QtPrivate::QSlotObject<void (WbTreeView::*)(QPoint const&), QtPrivate::List<QPoint const&>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*)
    (which=1, this_=0x55555c4cf040, r=0x55555c38d0c0, a=0x7fffffffcfb0, ret=0x0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:418
[... 8 lines cut ...]
#49 0x00007ffff615a412 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#50 0x00007ffff6162574 in QApplication::notify(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#51 0x00007ffff7a67808 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Core.so.5
[... cut ...]
#64 0x00007ffff7ac31cc in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#65 0x00007ffff7a6621a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#66 0x00007ffff7a6f1d3 in QCoreApplication::exec() () at /home/yann/webots/lib/webots/libQt5Core.so.5
#67 0x00005555559f0447 in WbGuiApplication::exec() (this=0x7fffffffdda0) at gui/WbGuiApplication.cpp:337
#68 0x0000555555bf832d in main(int, char**) (argc=1, argv=0x7fffffffdfe8) at gui/main.cpp:207
</pre>
</details>

**System**
 - Operating System: Debian 11

lukicdarkoo, 2021-06-15:
One note here. I couldn't reproduce the crash with the LensFlare in the Viewpoint (as it doesn't disappear after the reset), but I could reproduce it with a bounding object.

omichel, 2021-06-15:
I am closing the issue, feel free to re-open if needed.

PR #3174 (2021-06-15): Fix Context Menu Refresh
Fixes #3169. Webots now updates the context menu before it is shown in the scene tree.
closed 2021-06-15 by omichel