cyberbotics/webots #3170: Improve the axis system to allows 2D movement

ShuffleWire, 2021-06-12:
**Is your feature request related to a problem? Please describe.**
When I want to move my robot in Webots, I often have to change it position on the ground, but I don't want to change it's vertical position.
It would be nice to add some handles on the axis system of objects to be able to get a 2D movement.

**Describe the solution you'd like**
As an example, Blender does that nicely
[image]
However, in Webots, there are already handles to rotate object, which take place (and are useful, don't get me wrong)
FYI : Blender use a keyboard shortcut to toggle between displacement and rotation.

**Describe alternatives you've considered**
I'm often do two movements, one on each axis or I reset the vertical position after a 3D movement, but it's not very handy nor accurate.
I also could move to the top view, but I often don't want to change my current point of view on the scene, especially during demonstration to others people, for instance.

omichel, 2021-06-14:
The <kbd>shift</kbd>-click is meant for this: drag the robot on the ground without changing its height. Isn't it sufficient?

ShuffleWire, 2021-06-14:
Yes, this is completely fine, I've definitely missed that option ! I was thinking about extending this mechanism to the two other planes, but I agree that in Webots use case, it's not that relevant, moving robot in the horizontal plane is nice.
Maybe extend that to movement (ie translation+rotation setup) of any node, that could be useful when I'm placing one object relatively to the others.
Feel free to close if not relevant for you :)

closed 2021-06-14 by omichel