cyberbotics/webots #3178: python.exe" was not found.

juzhenghao, 2021-06-15:
**Documentation page(s) concerned**
E.g., for example https://cyberbotics.com/doc/reference/distancesensor#sonar-sensors

**What is the problem with the current documentation?**
A clear and concise description of what's wrong with the current documentation.

**Describe how you would fix it**
A clear and concise description of what you want to happen.

juzhenghao, 2021-06-15:
WARNING: "python.exe" was not found.
Webots requires Python version 3.9, 3.8, 3.7 or 2.7 (64 bit) from python.org in your current PATH.
To fix the problem, you should:
1. Check the Python command set in the Webots preferences.
2. Check the COMMAND set in the [python] section of the runtime.ini file of your controller program if any.
3. Fix your PATH environment variable to use the required Python 64 bit version (if available).

omichel, 2021-06-16:
I don't understand this issue. I am closing it. If needed, please open a new issue and provide a better description.

closed 2021-06-16 by omichel