cyberbotics/webots #3183: nao_demo crash

omichel, 2021-06-16:
**Steps to Reproduce**
1. Open nao_demo.wbt
2. Open the Nao proto in the scene tree
3. Select a color to replace the default color which is strangely -1 -1 -1
4. Reset the color field to its default value (-1 -1 -1) from the popup menu.
5. Click on the next field in the Nao proto: controller.
6. Webots crashes 😒

**System**
 - Operating System: Windows 10

BenjaminDeleze, 2021-06-16:
It also crashes on Ubuntu 20.04

omichel, 2021-06-16:
I believe that it is a bad idea to set a default color to `-1 -1 -1`, because in any case it will default to another color, red in this case. We should avoid that in addition to fixing the crash.

stefaniapedrazzi, 2021-06-23:
Just for note, this crash is not a regression.

The crash is specific to the `-1 -1 -1` value only in the sense that the field change is applied later than usual causing PROTO regeneration when it is not expected that results in accessing a field that has already been deleted.
I will now check how to fix properly fix this special case for all the editors that need it.

Then, `color -1 -1 -1` should no longer cause issues.
This trick allows to automatically set specific default color based on the robot version (v3.3 is grey, v4.0 is orange and v5.0 is red).
If we remove it, then we should apply the same default color independently from the version.

omichel, 2021-06-23:
>  If we remove it, then we should apply the same default color independently from the version.

That wouldn't be a big deal. Or we could consider the pure black color being automatically overridden.
According to the VRML97 specifications, it seems illegal to have a SFColor set to -1 -1 -1. Color components should be in the range 0.0 to 1.0, see https://www.web3d.org/documents/specifications/14772/V2.0/part1/fieldsRef.html

stefaniapedrazzi, 2021-06-23:
> According to the VRML97 specifications, it seems illegal to have a SFColor set to -1 -1 -1. Color components should be in the range 0.0 to 1.0, see https://www.web3d.org/documents/specifications/14772/V2.0/part1/fieldsRef.html

Yes. Actually [-1 -1 -1] is never used as color but immediately translated to a valid color to be applied to the instantiated nodes.
So it is just the definition in the PROTO parameter that doesn't follow the VRML standard.

I would rather set a fixed color, like the one for version 5.0 instead of adding additional complexity that could confuse users to overwrite a valid black color.

omichel, 2021-06-23:
> Yes. Actually [-1 -1 -1] is never used as color but immediately translated to a valid color to be applied to the instantiated nodes.
So it is just the definition in the PROTO parameter that doesn't follow the VRML standard.

Yes, the proto defines it as a SFColor, not a SFVec3f, so that seems wrong to me.

> I would rather set a fixed color, like the one for version 5.0 instead of adding additional complexity that could confuse users to overwrite a valid black color.

Yes, that's a good idea.

PR #3218 (2021-06-24): Fix crash due to invalid SFColor value
Fix #3183:
do not allow use of invalid SFColor values in PROTO parameters.
If invalid values are defined, they are clamped to the valid range [0.0, 1.0] and an error is printed in the console.
closed 2021-06-24 by stefaniapedrazzi