cyberbotics/webots #3207: Improve section including `wb_supervisor_node_export_string`

stefaniapedrazzi, 2021-06-21:
Currently the `wb_supervisor_node_export_string` is described in the same section as the `wb_supervisor_node_remove` function.
However it seems to be more closely related to the `wb_supervisor_field_import_mf/sf_node_from_string` rather than to the `wb_supervisor_node_remove`.
Would not be better to move it to the same section as `wb_supervisor_field_import_mf/sf_node_from_string`?
Because it doesn't seem to have much to do with the function that removes nodes.

https://www.cyberbotics.com/doc/reference/supervisor?version=develop#wb_supervisor_node_export_string

lukicdarkoo, 2021-06-21:
I cannot recall why I put it there but your suggestion totally makes sense

stefaniapedrazzi, 2021-06-21:
Ok. So then I can take care of moving the description.

closed 2021-06-30 by stefaniapedrazzi